shadow-1/yi-hack-v3 #66: Yi Dome Camera with 43CNYxxxx

minfui, 2017-10-20:
my Yi Dome Camera is come with serial 43CNYxxxx
trying many times but with the firmware on SD card
but it seem like by pass, & run into "waiting for connection .... "
is this model different from others ?
thank you very much

minfui, 2017-10-20:
by the way, I am using MacOS, are this effect the firmware flashing?
but I did format the card in FAT32 in Disk Utility

shadow-1, 2017-10-20:
@minfui 
From what you have described. It is a little hard working out exactly where you are having issues. I am not sure whether 43CN firmware is different to the earlier versions. It is unlikely as Xiaomi appear not to have a separate firmware for the newer Yi Dome cameras.

Did the yellow light blink on boot when flashing the firmware?

Does the blue light blink when the camera tries to connect to your WiFi network? Does the blue light turn solid at any time?

minfui, 2017-10-20:
first of all, thank you very much for the fast reply

1. when I put in the sd card & power up, the light change to blue for once ( I think its mean that its detect the sd card), but the camera with turn the right side, after a few second, its move back to front & "welcome ..... waiting for connection", now the light is yellow blinking, but the connection announcement is still keeping on continue, again & again

2. if I continue doing the step of connection, scanning QR code, actually it did connect to my wifi (blue light on), but failed in pairing & say its only can use within China (that why I want to flash it 😂)

even I use the official firmware to put on the sd card, same happen, its just like skip to waiting for connection step 😭 (actually I start to suspect the way I flashing the firmware is correct or not 😓)

minfui, 2017-10-21:
Update:
its sd card problem, its work when I change to another sd card
very sorry to disturb u, & thank you very much for your help

shadow-1, 2017-10-21:
@minfui 
Glad that you worked it out. I have had issues with some particular microSD cards as well in the past.

closed 2017-10-21 by shadow-1