shadow-1/yi-hack-v3 #67: Does this firmware supports ONVIF?

DuckY1987, 2017-10-24:
I would like to use the camera, only with my Synology NAS, does this firmware has ONVIF feature?

shadow-1, 2017-10-24:
@DuckY1987 
This firmware does not support ONVIF yet. At the moment, this firmware only supports the official Xiaomi smartphone app and PC software.

I see ONVIF support as a natural progression after RTSP support. RTSP is a sought after feature, however it is only a video/audio streaming protocol, it will not facilitate the use of the PTZ functions of the Yi Dome cameras.

I initially plan on releasing a firmware compatible with the original Yi Home camera, after that I plan to develop RTSP support and perhaps after that I will look into developing ONVIF support (mainly to allow PTZ functionality without the Xiaomi app).

sdho2004, 2018-03-10:
@DuckY1987 
I know it's not ideal since you cannot use SurveillanceStation  to manage the cam but I've been happy with using Yi Home app and automatically backing up *mp4 files using lftp script to my Synology.

-sdho

sdho2004, 2018-04-07:
@felipe-vvoosh ,

You will need to create a .sh file like the following: "backup.sh" in the /volume1/backup/ directory.  You can use any directory under /volume1/.

The content of backup.sh file:
`#!/bin/bash
login="root"
pass=""
host="192.168.1.21" #IP address of your Camera
local_dir="/tmp/sd/record" #Local camera directory
remote_dir="/volume1/backup/" #DSMStation directory

trap "rm -f /tmp/backup.lock" SIGINT SIGTERM
if [ -e /tmp/backup.lock ]
then
echo "Backup is running already."
exit 1
else
touch /tmp/backup.lock
lftp -p 21 -u $login,$pass ftp://$host << EOF
mirror -c --log=/volume1/backup/backup.log $local_dir $remote_dir
quit
EOF
rm -f /tmp/backup.lock
trap - SIGINT SIGTERM
exit 0
fi`

Make backup.sh file executable with chmod 755 command and add it to your DSM Station task scheduler.  I have the task running daily for every 30 min with the first run time at 00 and the last run time of 23:30.  You can make it more frequent if you'd like.

-sdho

delfinogrisu, 2019-02-19:
hi, I would be very interested in this project, you could also have the support of the convif, the post seems very old and I wanted to know if there were any news thanks a lot

shadow-1, 2019-02-20:
@delfinogrisu 
Check out yi-hack-v4, RTSP support is a work in progress on that project. ONVIF support is essentially an extension to RTSP. Very handy for PTZ cameras.

delfinogrisu, 2019-04-18:
hello shadow-1 I contact you with the hope that you can help me to resume the project and solve some problems and create something more than what you've already done and has already done https://github.com/niclet/yi -hack-v2.
it is possible to have complete sdk and toolchain for the related version 2 in order to continue the development and creation of rtsp etc. since it is doing crypto with the yi-hack v4 project to which I have already explained the problem and I would also like to integrate the ip camera v2 ...
  you don't know where you can procure full sdk with yi camera home 1080p v2 toolchain ??
I invite you to enter the channel and have a chat and tell you about the progress I've already made with this room ;-)
https://discord.gg/upPsFWZ

shadow-1, 2019-04-21:
@delfinogrisu 
Version 2 of the camera uses a completely different chipset to yi-hack-v3.
I think version 2 of the camera utilises an Ambarella S2LM chipset.

A quick search online found that the SDK for this chipset is difficult to come by. I could not find a link.

I think the first course of action is to get access to the cameras serial interface and determine precisely which chipset is used on this camera.

delfinogrisu, 2019-04-21:
@shadow-1 thanks for the answer, I know it is different hardware but knowing that you have experience in the field with version 1 I think it can make a great contribution and help to the project. if you can give me a hand we will continue to carry out this project too -
```
Machine: Ambarella S2L (Flattened Device Tree), model: Ambarella S2LM Kiwi Board
Linux version 3.10.73 (lisong@ubuntu) (gcc version 4.8.2 20130902 (prerelease) (crosstool-NG - Ambarella Linaro Multilib GCC [CortexA9 & ARMv6k] 2013.09) ) #1 PREEMPT Thu Jul 27 17:04:22 CST 2017
```
thank you so much for your precious help !!

delfinogrisu, 2019-04-21:
```
on the chip:
AMBARELLA S2L-M-A1-RH
A1602
N9583-AN4
1N1 
S2L 33M
```

martindevnow, 2019-05-23:
@sdho2004 
Does this work out of the box? Do I need to install some firmware to the camera first?
I tried running it, but I'm getting connection refused. 

I've got the Yi Cloud Home Camera and running firmware 1.9.3.0E_201812141519

sdho2004, 2019-05-24:
@martindevnow yes, you will need to install yi-hack-v3 firmware from shadow-1 first.  Please follow his README from the main page.  FYI, Or you can try out yi-hack-v4 by theCrypt0 which I believe has rtsp capabilities so you may not need to FTP anymore.