shadow-1/yi-hack-v3 #68: downgrade back to the standard firmware

xupipi88, 2017-10-24:
Is it possible to go back the default firmware?
I have downloaded the 2.1.0.0C_201710091352home_y20m from YI homepage but it is not upgraded with this firmware? Is there any limitation with hack-v3 to avoid such upgrade?

The reason for it is that I cannot pass the last step to find a working proxy with this HACK-V3 and want to give the camera back to my chinese friend who lives in China and I am afraid it will not work with the HACK-V3 in china.

Thanks for your answer!

shadow-1, 2017-10-24:
@xupipi88 
Yes it is possible to go back to default firmware. I have provided instructions on the main page. Have a look at the following section on the main page: https://github.com/shadow-1/yi-hack-v3#going-back-to-stock-firmware

I always want to provide an option to opt-back to the default/stock firmware. A lot of other firmwares do not provide such an option and sometimes unexpected things happen when the firmware modifies files on the filesystem without providing a way to undo the changes.

I have not had the time to test my firmware lately. I have developed the feature to automatically find and test proxy servers. However as far as I know, there are not as many working proxies from the source that I find proxy servers lately.

The camera will work perfectly in Mainland China with yi-hack-v3 installed. However ProxyChains-ng should be disabled as it is not required and would be an unnecessary nuisance.

xupipi88, 2017-10-24:
thanks for your answer.
I am not sure where you are from China. 
It is already to buy the europe verison YI in china with fair price. If you need it, I can send you the link.

Thanks for all you do and I hate XIAOMI YI company :-)

rdhood, 2017-10-24:
This might be important to more than just your user. I tried "upgrading"  my 1080p camera when the Yi app requested (to the 2.1.0.0C_2017xxxxxx firmware) and they never recovered. I had to go back to the "B" August revision.  Just a heads up that the "C" firmware might be flaky for you...

shadow-1, 2017-10-25:
@xupipi88 
I am not from China. I developed this hack primarily to allow the use of region locked cameras outside of Mainland China using the official Xiaomi app.

@rdhood 
Thank you for this information. I will have to examine the firmware to see what Xiaomi changed in this firmware release and what could have gone wrong.

shadow-1, 2017-10-29:
@rdhood 
I have extracted the new 2.1.0.0C firmware and did not find it too different from the previous releases.

Can you describe what you mean by it never recovered after the upgrade?
Does the web server still work? What about Telnet or FTP? Did you try restarting the camera again? Did you go back to 2.1.0.0B by flashing yi-hack-v3 again or did you use the official firmware file?

I do not have this camera so it is a bit hard for me to pinpoint what the issue could be.

helijoop, 2017-12-16:
hi,
I`m new in YI Dome camera. I need your idea about this great job for adding features to default firmware, do you think is it possible xiaomi finds which camera has hacked firmware and try to lock it ? 
another question, is there any android app which cover all options for camera that we replace that app with original app !  it is just because we move from xiaomi server  and be standalone with another third-party app .
thank you

shadow-1, 2017-12-17:
@helijoop 
It is very unlikely that Xiaomi will block cameras which have the hack installed.

However there is a chance that Xiaomi will find a way to get around the region lock workaround that I implemented. If this happens, cameras which are region locked to Mainland China may not work.

At the moment, there is no app which can replace the official Xiaomi app. However when RTSP support gets implemented, this will open the path for third-party apps.

helijoop, 2017-12-17:
Thank you for quick answer and give better view on using your great project hacked firmware. 
there are beta released and final released firmware for cameras, what is your suggestions for installing beta or final release for YI Dome camera. we used to use beta released software/firmware for devices  with a risk of stability. or beta versions are only small improvement of previous stable version without any risk of stability!

Thank you

shadow-1, 2017-12-17:
@helijoop
There are significant improvements with the beta release. It has been out for a while and there appear to be very few issues with it.

That would be my recommendation.

poureslami, 2022-02-13:
I upgrade my Chinese camera yesterday and now it doesn't work.
I tried to downgrade it many times but it still doesn't work.