shadow-1/yi-hack-v3 #70: Automatically send to FTP

joaoasilva, 2017-10-26:
Hi,

Is it possible to automatically send the recorded videos to FTP?

Thanks.

shadow-1, 2017-10-26:
@joaoasilva 
Yes it is possible. Some scripts have been developed by others to fetch recorded video data automatically.

Most people write a script that runs on a small board like Raspberry Pi or potentially their router to automatically fetch the recorded videos and copy them to their NAS or cloud storage (Google Drive, DropBox etc).

AppaloosaTech, 2017-10-28:
@joaoasilva have you found a script that fetch recorded video data automatically to ftp? I'm searching such a script to save the data on my router.

@shadow-1 could that be an alternative/easier solution to stream videos as via rtsp?

shadow-1, 2017-10-29:
@AppaloosaTech 
Two people have successfully written scripts to download recorded video files automatically using FTP. Unfortunately they did not share the scripts in the thread. Link: https://github.com/shadow-1/yi-hack-v3/issues/13

RTSP support is a work in progress at the moment.

mrgoodfellow, 2019-02-03:
I am working on this functionality.  My concept is to have a .sh script that listens to the /tmp/sd/record directory for file changes and then automatically pushes those files to a sftp server.

https://www.computerhope.com/unix/sftp.htm

I'm working on getting this setup.  I had been trying to use the ftpput command and was getting some errors.  Any resources on setting up sftp or automating this process would be awesome!

https://askubuntu.com/questions/758640/how-to-automatically-sync-the-contents-of-a-local-folder-with-the-contents-of-a

Catfriend1, 2020-07-12:
Frp puah is supported in https://github.com/roleoroleo/yi-hack-Allwinner since v0.1.7