shadow-1/yi-hack-v3 #71: new firmware support. 2.1.0.0C

sfordinc, 2017-10-28:
А new version of the firmware was released - 2.1.0.0С_201710091352
Please rebuild the release with this firmware.

shadow-1, 2017-10-29:
@sfordinc 
I have heard from another user that upgrading the firmware to 2.1.0.0C does not work properly (I have implemented a script which should recover yi-hack-v3 after a Xiaomi firmware update).

Can you describe what happens when you upgrade to 2.1.0.0C?
Does the web server still work? What about Telnet or FTP? Did you try restarting the camera again? Can you downgrade back to 2.1.0.0B using the official firmware file or do you have to flash yi-hack-v3 again?

I do not have this camera so it is a challenge for me to pinpoint what the issue could be.

sfordinc, 2017-10-29:
@shadow
After updating the firmware to 2.1.0.0C from 2.1.0.0A, the camera itself is working fine. And it is this version after the rollout of yi-hack-v3 that it is installed - 2.1.0.0A, and not 2.1.0.0B. If you still update the firmware via the mobile application, then telnet, ftp, web services stop working. Although maybe for the first time I did not use the latest version of the hack, but I would not want to repeat the update to the new firmware. It's very dreary.
I would like to look at the script, which restores yi-hack-v3 after the firmware update Xiaomi.

Another little question for off topic. Do not restart rtspsvr.
After launching it produces such a log and the 554 port does not open to listen:
/ home / app # ./rtspsvr

Set 0 UID = DF899HNEJMAFVNPPWFXT
[59f4e3fc] [RTSP] main: renice -n -1 -p 1379 ok!
[59f4e3fc] [RTSP] call get_sharemem return ok!
[59f4e3fd] [RTSP] LwlibOpenMsgQueue: mq_open ok!
[59f4e3fd] [RTSP] rtspsvr_init ok!
V
[59f4e3fd] [RTSP] chn 3 cmd: renice -n 1 -p 1379 ok!
[59f4e3fe] [RTSP] chn 1 cmd: renice -n 1 -p 1379 ok!
[59f4e3fe] [RTSP] chn 0 cmd: renice -n 1 -p 1379 ok! V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V V
[RTSP] mq_send WDG_FEED_TICK error: Resource temporarily unavailable
V
[RTSP] mq_send WDG_FEED_TICK error: Resource temporarily unavailable
....

shadow-1, 2017-10-29:
@sfordinc 
Ah that makes sense. Only the latest version of yi-hack-v3 is able to restore itself after a Xiaomi firmware update (version 0.1.4-beta2 onwards). Prior to this version, Xiaomi updates would break yi-hack-v3.

The script which restores yi-hack-v3 is this one: https://github.com/shadow-1/yi-hack-v3/blob/master/src/home/yi-hack-v3/script/system_init.sh

Which RTSP implementation are you using? I have not provided an implementation and it is not included with the Xiaomi firmware updates.

sfordinc, 2017-10-29:
@shadow
Thanks for the link to the script. I'll try to use it somehow.
Take the files with rtsp-server by reference. They were used in the previous camera yi-113 720p.
https://yadi.sk/d/nBTmkeoo3PCJTG

shadow-1, 2017-10-29:
@sfordinc
Ah ok. Unfortunately RTSP from the old camera will not work with the newer cameras.

It is a shame that the source code for that implementation has not been released. It should be relatively easy to adapt that RTSP implementation to the newer cameras.

closed 2017-11-18 by shadow-1