shadow-1/yi-hack-v3 #73: connection to this camera exceeded allowable limit. (-3029)

fier98, 2017-10-30:
Hi..
Thanks shadow-1 for your very helpful works.
After I installed yi-hack-v3, my Yi Dome international version camera worked nicely for 3 days. But, later I have problem with it which was failed to connect. The message pop up on YiHome app screen "Network connection failed. Please check network settings and reconnect it (-42)" and "connection to this camera exceeded allowable limit. (-3029)". In fact, I can ping the proxy's ip address from the camera.

I tried to fix it by resetting and re-pairing again thru YiHome app but the problem cameback with the same messages.

Any ideas why this thing happen?

shadow-1, 2017-10-30:
@fier98 
Can you post the output from the Test Config webpage. The Test Config page does a few tests and is more comprehensive than just a ping. Being able to ping the proxy server address does not guarantee that it is working.

When you get this problem. Are you able to access the yi-hack-v3 web interface?

fier98, 2017-10-30:
I still can access the yi-hack3 interface. This is the test config output:

[image]

shadow-1, 2017-11-17:
@fier98 
This is strange. According to the screenshot, the proxy server works. There is the very off occasion that a proxy server passes the test but still won't work. I have incorporated another check which will be part of the next firmware release.

---

I just realised, you are using Yi Dome international version. With International version of the camera. You do not need ProxyChains-ng as it is not region locked. You can disable ProxyChains-ng on the System Config page.