shadow-1/yi-hack-v3 #74: help plz ProxyChains-ng error

plutosherry, 2017-10-30:
ProxyChains-ng Download Proxy List error log :

wget -O /tmp/temp_proxy_list.txt https://freevpn.ninja/free-proxy/txt 

Connecting to freevpn.ninja (173.208.176.114:443)
wget: can't connect to remote host (173.208.176.114): Connection timed out

shadow-1, 2017-10-31:
@plutosherry 
This error message indicates that the camera could not access the online page with the list of proxy servers.
What happens when you try to visit https://freevpn.ninja/free-proxy/txt from your computer?

plutosherry, 2017-11-08:
Unable to access the ‘https://freevpn.ninja/free-proxy/txt’ site
Cannot resolve DNS

shadow-1, 2017-11-08:
@plutosherry 
I can confirm that the web page works. Unfortunately it appears that your ISP may be blocking access to https://freevpn.ninja (there is a small chance that it is just blocked on your router).

Maybe try visiting the webpage from another network to confirm. For example through mobile network.

If nothing helps, you will need to find a working proxy server from Mainland China manually.

plutosherry, 2017-11-17:
@shadow-1 
Https://freevpn.ninja can not access, the use of VPN is not possible, it may be the problem of ISP operators, if you can not successfully open the ProxyChains-ng, then it is not able to use it?

shadow-1, 2017-11-17:
@plutosherry 
It is still possible to use proxychains-ng. However you will have to find working proxy servers from Mainland China yourself. Some useful websites include:
www.xroxy.com
http://www.gatherproxy.com/proxylist/country/?c=China
http://spys.one/free-proxy-list/CN/

These can be added to the end of your configuration file on the web interface.
Once populated, you can use the "Test Proxy List" feature to test them and only keep the ones that work.

plutosherry, 2017-11-19:
I've manually added the proxy server address of the camera to web in config, and added it to hassio's configuration.yaml.... 

camera:
  - platform: yi
    name: yi Camera
    host: '192.168.1.100'
    password: 'password'
    path: /home/camera/feed
    ffmpeg_arguments: '-vf scale=800:450'

However, the state of the camera in the hassio is 'idel'

shadow-1, 2017-11-19:
@plutosherry 
I suggest testing all the proxy servers using the "Test Proxy List" feature in the web interface and only keep the proxy servers that work (many listed online do not work).

I then suggest making sure that the camera works with the Yi Home app before trying other things.

Can you post a screen shot of your ProxyChains-ng configuration as well as the results when using the "Test Proxy List" page?

plutosherry, 2017-11-19:
The following is screenshot information：
[image]
[image]
[image]

plutosherry, 2017-11-19:
If I open the ProxyChains-ng then the YI home app will stay in the video buffer 25%, if I don't open proxy the YI home app can display the normal video content. However hassio can not display the video anyway, why?

shadow-1, 2017-11-20:
@plutosherry 
I am a little confused. Do you have the International version of the camera or the Chinese version?
Only the Chinese version requires ProxyChains-ng as Xiaomi have implemented a region lock which only allows the Chinese version to be used in Mainland China. The International version of the camera does not require ProxyChains-ng.

---

I am not familiar with the Home Assistant plugin. My understanding is that it fetches the motion detection video recordings from the camera using FTP.

Can you confirm that you are using the microSD card when using Home Assistant? The microSD card stores the video recordings which Home Assistant fetches.

From the Home Assistant plugin guide, it says that the system password needs to be changed from the default (which is blank). Can you confirm that you have done this?

plutosherry, 2017-12-15:
@shadow-1 Yes, I use the Chinese version, so I need to enable ProxyChains-ng.
After finding the appropriate proxy server address, And now it is working but very unstable, i doubted that it may be the network problem of proxy services.
Thank you for the reply

shadow-1, 2017-12-17:
@plutosherry 
This must be very frustrating for you! I can confirm that the proxy server is only used for authentication on the Xiaomi server. Video and audio data does not go through the proxy server. There must be some other issue on your network (maybe even your Internet Service Provider) which is causing stability problems.

What is happening that makes it unstable?

LaFed, 2018-01-04:
hi.... i manually keyed in the proxy and managed to find 5 working ones... in the instructions it says connect from smartphone, since pairing is not successful in the beginning, how can I connect with my phone? do i reset the camera and pair again? I tried reseting and pairing still not successful even with working proxy server

plutosherry, 2018-01-07:
@LaFed 
As you can see, after adding Yi dome to homeassistant, the video is not smooth and not real-time. Of course, I think the reason is not on the firmware, but because the Chinese version of Yi dome connects to the server.

plutosherry, 2018-01-07:
@shadow-1 
First of all, thank you for your busy schedule to reply to my message
Because of the recent things are much more, so the problem has not been solved.
I don't know if I can bypass the authentication server through a LAN direct access to Yi dome? I mean to play the local video stream directly through the RTSP protocol

LaFed, 2018-01-07:
@plutosherry I was having problem initially of showing the cam in HA. Managed to solve that which I'm happy. When you say video isnt smooth, do u mean in HA or Mi app? for HA, it's understandable as it grab a still from recorded mp4 which is generated every minute. I have other yi home that I've already hacked to enable rtsp which is showing live (still pic every 10 second I think) which gives better result. Hopefully @shadow-1 will be able to add rtsp without killing mi app connection in the future. Looking forward to it

gm1231s, 2018-01-08:
I am using the black cn 1080 Dome with the cn app. Added a few proxies and tried Test Config to be working.
But somehow during pairing, it kept prompting in mandarin "pairing timeout".
I tried using the latest and older cn Yi home app but I still faced the pairing timeout issue.

Can anyone guide me what else I can do ?

Camera     Yi 1080p Dome (45CN)
Manufacture date : Oct 2017
Base Firmware     1.9.2.0F_201708041132
yi-hack-v3 Firmware     0.1.5
App version : 2.10.19.1_20171213
App version : 2.10.19.1_20170512

[image]

[image]

shadow-1, 2018-01-09:
@gm1231s 
The first thing I would do is try using Socks4 or Socks5 proxy servers rather than HTTP. I have had far less success with HTTP proxy servers.

chilskater, 2018-03-19:
Anyone working proxies??I try 20 and give up

plutosherry, 2018-03-20:
i am already give up

发自我的小米手机

YYL1129, 2020-10-10:
> The following is screenshot information：
> [image]
> [image]
> [image]

how to access to this link? i am able to detect the camera ip address but unable to open this page. 
please help.