shadow-1/yi-hack-v3 #76: camera continue to load at 25% and timed out

kohorok, 2017-11-02:
It has been working fine with your proxy setting, but suddenly today i was unable to connect to the camera.
As usual, camera will show 5%...25% and connected, but today It stopped at the 25% establishing a secure connection...and after couple of minutes it timed out.
It looks like it is trying to connect, but it always failed.

Fyi, I have not update the firmware that always prompt when i launch the app.

Any idea?

shadow-1, 2017-11-03:
@kohorok 
I would try the Test Config and Test Proxy List features on the hack webpage.

What this sounds like is that the configured proxy servers are not working and the camera has performed a region check. In this case, the connection to the Xiaomi server times out and the camera can no longer to accessed through the app.

kohorok, 2017-11-05:
@shadow-1 
Hi,
Would you recommend update of firmware?

shadow-1, 2017-11-08:
@kohorok 
Yes I would recommend updating to the latest version of yi-hack-v3 and the Xiaomi firmware.