shadow-1/yi-hack-v3 #77: Static orange light

joaoasilva, 2017-11-02:
So, I was stupid enough to install the home and rootfs from version 17cn to a 22cn and now my 22cn doesn't start, static orange light.
I've already tried to revert with the firmwares available here: https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr
But nothing... it blinks for some time but then it stops again and that's it.
I've also tried the original firmware from the oficial website but nothing... any ideas?

Thanks.

Regards,

shadow-1, 2017-11-03:
@joaoasilva 
The reason why your efforts in recovering the firmware have been unsuccessful is because the rootfs partition between the Yi Home and the Yi Home 17CN cameras is completely different.

I will develop a set of recovery images compatible with the original Yi Home.

shadow-1, 2017-11-03:
@joaoasilva 
Try the new recovery image I created for the Yi Home:
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

It is in the directory; `yi-hack-v3->Recovery->Yi Home`

Let me know if it works properly or if you are still having issues. I based this recovery image on an old model of the Yi Home camera. From memory it was the 13CN version (model before Xiaomi implemented region locking on their cameras).

joaosilvareed, 2017-11-03:
Thank you so much @shadow-1 !! Worked perfectly!
Do you have any ETA to support this camera?
Just want to get HomeAssistant working with the new plugin :)

Thanks once again!

joaoasilva, 2017-11-03:
Thank you so much @shadow-1 !! Worked perfectly!
Do you have any ETA to support this camera?
Just want to get HomeAssistant working with the new plugin :)

Thanks once again!

shadow-1, 2017-11-08:
@joaoasilva 
Glad everything worked.
I may have a pre-release for the original Yi Home coming soon.

closed 2017-11-13 by shadow-1