shadow-1/yi-hack-v3 #78: Yi Home 1080p not updating

slacker247, 2017-11-04:
I'm not sure if I'm doing something wrong or the camera just isn't accepting the update.  I've copied the home_y20 and the rootfs_y20 to a 32fat formatted 8gb sd and copied these 2 files over.  Booting up with this sd card.  The camera shows the yellow steady light the, switches to off, which is the setting I have it at.  Pulling the sd card after this, the card has .tmpCatch, YH1080P_Record, YH1080P_Record_sub, .ants_sd_index.  I'm assuming this is the standard initialization of an sd card.  The 2 files home_y20 and rootfs_y20 are still there.

shadow-1, 2017-11-08:
@slacker247 
Make sure that the microSD card is formatted to FAT32. The camera works with exFAT formatted cards after it boots up. However the firmware update occurs before the camera boots the OS and it only supports FAT32 at this time.

I would also suggest trying a different microSD card. I have occasionally had issues with some cards which could not be resolved no matter what I tried.

slacker247, 2017-11-14:
I've tried a bunch of different sd cards and they are all doing the same thing.  SanDisk, Kingston, no name.

shadow-1, 2017-11-17:
@slacker247 
Does the camera blink a yellow light when booting up with a firmware image? Try flashing an older version before the latest version. I have had the most success with small SD cards. 16gb or less.

slacker247, 2017-12-04:
Yellow light stays solid, then turns off.  Cycles the shutter, then is operating stock firmware.
Tried all versions of y, I also tried v and h just to see.  same behavior.
Tried hard reset and now the process of turning it on is, solid yellow light, audio "Camera starting up", blinking yellow light, audio "waiting to connect", 
I step through the connection steps, blue light and now it's connected.  Tried connecting to it via the web browser, "connection refused"
Version number of the firmware on the camera - 2.1.1_20170606142800

shadow-1, 2017-12-05:
@slacker247 
I believe the issue is that the firmware is not flashing on your camera for some reason. In my experience this is usually because the microSD card is not formatted properly with FAT32 filesystem. This is usually an issue with larger microSD cards which default to being formatted to exFAT which will not work. However it can be an issue with any microSD card.

Try formatting the microSD card with the official SD Card Formatter tool. It is available here: https://www.sdcard.org/downloads/formatter_4/

shadow-1, 2017-12-05:
@slacker247 
I just realised what the issue is! You mentioned your firmware version is 2.1.1_20170606142800.
I checked the Yi website and confirmed that this firmware version is specific to the Yi Home 2.

Unfortunately this firmware does not support Yi Home 2 camera as it uses a completely different chipset to the other cameras I support.
The Yi Home 2 makes use of the Ambarella S2LM chip.
This firmware only supports cameras based on the HiSilicon Hi3518eV200 chipset.

The firmware linked below supports the Yi Home 2 which you can try.
https://github.com/niclet/yi-hack-v2