shadow-1/yi-hack-v3 #79: Any plans to add live view and controls to the web server?

lazibayer, 2017-11-09:
I love the camera but I wish I can use it at places without ANY internet connection. I have an old Fujikam that allows me to do that by simply navigating a browser to its IP address.

shadow-1, 2017-11-10:
@lazibayer 
I do not plan on developing this feature any time soon.

However I am planning on developing RTSP support which will allow use at places without an Internet connection. However not through the web server but through a program running on a PC or through a smartphone app which will not require Internet access.

lazibayer, 2017-11-10:
That's good news! Thanks for your continuous effort!

alexgorbatchev, 2017-11-23:
May I suggest a similar feature that could be "easier" to implement. A static JPEG url that could be accessed and which is updated every N seconds. This would integrate very well with home automation systems like openHAB which allow to display URL images on the monitors and refetch them every N seconds.

Thoughts?

alexgorbatchev, 2017-11-23:
Also, this project allows RTSP streaming https://github.com/xmflsct/yi-hack-1080p

Ben47, 2017-11-27:
Hi, I have a camera yi dome 720p international. I put the firmware yi-hack-v3.
How can I access the video stream of my camera on a browser?
(ex -> http: // ip_camera :?) What are the internal and external TCP port numbers to use?
thank you in advance
Benoit
[image]

lazibayer, 2017-11-28:
> 

Also, this project allows RTSP streaming https://github.com/xmflsct/yi-hack-1080p
--

I check that project and it seems that RTSP and Yi Home access are mutually exclusive. Will it be the same case for this project as well?

kruschman, 2017-11-28:
It looks like that project is only for the 1080p version. Does anyone know if it will work with the 720p 27US version since they apparently have the same image sensor?

shadow-1, 2017-11-28:
@alexgorbatchev 
I will have to investigate whether an updating webpage will be easier to implement than RTSP.
I know of this project as I have had some input into it.

@Ben47 
Unfortunately only the official Xiaomi apps are supported at this time (Android, iOS and Windows).

@lazibayer 
I do not plan to make RTSP and Yi Home access mutually exclusive. However I will have to see how I go.

@kruschman 
Yes this other project is specifically for the Yi 1080p Home camera. As far as I am aware, the 720p 27US version of the camera uses a different image sensor. Therefore to make this preliminary RTSP implementation work on the 720p cameras, it needs to be recompiled against the correct image sensor library.

dvv, 2017-11-29:
+1 for managing to obtain the static JPEG image.
This is beacuse if it _is available somehow_, it can be sent with periodic scripts, accessed via ftp polling, busybox's (more fat) httpd applet, hand-made poorman shell-based http server, used as the source for mjpeg-streamer's input_file.so etc etc etc you name it.

Just FYI, in motion detection mode `/tmp/motion.jpg` becomes available at motion start.

shadow-1, 2017-11-30:
@dvv 
Thank you for the information. I wonder if the JPEG image gets updated regularly or whether it gets created once per motion event.

dvv, 2017-11-30:
Once. It gets rid after several seconds and I had no chances to play with it more thoroughly. I believe it's sent to cloud then quickly removed.

I managed to (seemingly) robustly catch motion alarm start by the next script (by heart, no camera at hands):
```sh
tail -F /tmp/log.txt | awk '
/: got new motion start/ {
  print "ALARM+"
  system("<START HERE A SHELL COMMAND TO NOTIFY OF ALARM HAS BEGUN>")
}
/: got new motion stop/ {
  print "ALARM-"
  system("<START HERE A SHELL COMMAND TO NOTIFY OF ALARM HAS ENDED>")
}
'
```

Moreover, there are some debug info in `/tmp/log.txt` containing URLs the camera sends to the cloud -- may be it's worth to catch-proxy them for some kind of data or further kludges.

nicolasplanchon, 2017-12-04:
@dvv could you tell us a little bit more regarding the way you implement your script please ? :)

dvv, 2017-12-07:
I added a script `/home/yi-hack-v3/script/alarm.sh`:

```sh
#!/bin/sh

exec tail -F /tmp/log.txt | awk '
/: got a new motion start/ {
  print "ALARM+"
  system("wget -q --spider \"http://192.168.43.1:8765/sms?phone=XXXXXXXXXX&text=cam4\"")
}
/: got a new motion stop/ {
  print "ALARM-"
}
'
```
and made it called from `/home/yi-hack-v3/script/system.sh`:

```sh
if [ -f "/home/yi-hack-v3/script/alarm.sh" ]; then
        sh /home/yi-hack-v3/script/alarm.sh &
fi
```

The camera is WiFi-ed via an android phone in WiFi hotspot mode, so the phone's IP is known to be `192.168.43.1`.

I installed tasker on that phone with TNES plugin which listens to port `8765`. I added the task to listen to incoming HTTP GET requests to `/sms` endpoint, extract `phone` and `text` variables from the querystring and send SMS.

It just does work )

shadow-1, 2017-12-08:
@dvv
The work you have done is impressive. Being able to send a sms message upon motion detection is quite an achievement.

dvv, 2017-12-09:
Right, with https-powered `wget` in hands and robust detection of start of motion we can do very much.
I would post to telegram:
```sh
wget -q --spider 'https://api.telegram.org/bot<APIKEY>/sendMessage?chat_id=<CHAT>&text=ALARM!'
```

dvv, 2017-12-09:
The `log.txt` parser evolved:

```sh
#!/bin/sh

# Copyright 2017 Vladimir Dronnikov
# GPL

exec tail -F /tmp/log.txt | awk '
/: got a new motion start/ {
  # motion just started
  print "ALARM+"
  #system("wget -q --spider \"http://192.168.43.1:8765/sms?phone=9190504182&text=cam4\"")
}
/: cmd=\/home\/app\/cloudAPI -c 411 .* -filename \/tmp\/motion\.jpg\.crypt/ {
  # photo ready
  print "PHOTO"
  system("cp /tmp/motion.jpg /tmp/temp.jpg ; nc 192.168.34.10 1884 < /tmp/temp.jpg &")
}
/: cmd=\/home\/app\/cloudAPI -c 411 .* -filename \/tmp\/motion\.mp4\.crypt/ {
  # video ready
  print "VIDEO"
  system("cp /tmp/motion.mp4 /tmp/temp.mp4 ; nc 192.168.34.10 1885 < /tmp/temp.mp4 &")
}
/: got a new motion stop/ {
  # motion stopped, cleanup
  print "ALARM-"
  system("rm -f /tmp/temp.*")
}
/_____Gamma test for table 4__________/ {
  # daylight mode
}
/_____Gamma test for table 2__________/ {
  # night mode
}
'
```

I do miss https-enabled `curl` for uploading photo and video and replaced it with `nc` + some helper at another host in local net so far.

@shadow-1 couldn't you provide us with `curl`?

dvv, 2017-12-09:
I finally found that curl and now the script looks like:

```sh
#!/bin/sh

# Copyright 2017 Vladimir Dronnikov
# GPL

TELEGRAM_BOT_TOKEN=XXX
TELEGRAM_CHAT_ID=YYY

# NB: curl from https://github.com/utya1988/yi-hack/tree/master/sd/test/curl

exec tail -F /tmp/log.txt | awk -v TELEGRAM_BOT_TOKEN="${TELEGRAM_BOT_TOKEN}" -v TELEGRAM_CHAT_ID="${TELEGRAM_CHAT_ID}" '
# TODO: urlencode text
function sendText(text) {
  system("curl -k -q --data \"chat_id=" TELEGRAM_CHAT_ID "&text=" text "\" \"https://api.telegram.org/bot" TELEGRAM_BOT_TOKEN "/sendMessage\"")
}
function sendPhoto(file, text) {
  system("curl -k -q -F photo=@" file " \"https://api.telegram.org/bot" TELEGRAM_BOT_TOKEN "/sendPhoto?chat_id=" TELEGRAM_CHAT_ID "&caption=" text "\"")
}
function sendVideo(file, text) {
  system("curl -k -q -F video=@" file " \"https://api.telegram.org/bot" TELEGRAM_BOT_TOKEN "/sendVideo?chat_id=" TELEGRAM_CHAT_ID "&caption=" text "\"")
}
/: got a new motion start/ {
  # motion just started
  print "ALARM+"
  sendText("ALARM+")
}
/: cmd=\/home\/app\/cloudAPI -c 411 .* -filename \/tmp\/motion\.jpg\.crypt/ {
  # photo ready
  print "PHOTO"
  system("cp /tmp/motion.jpg /tmp/temp.jpg")
  sendPhoto("/tmp/temp.jpg", "PHOTO")
}
/: cmd=\/home\/app\/cloudAPI -c 411 .* -filename \/tmp\/motion\.mp4\.crypt/ {
  # video ready
  print "VIDEO"
  system("cp /tmp/motion.mp4 /tmp/temp.mp4")
  sendVideo("/tmp/temp.mp4", "VIDEO")
}
/: got a new motion stop/ {
  # motion stopped, cleanup
  print "ALARM-"
  sendText("ALARM-")
  system("rm -f /tmp/temp.*")
}
/_____Gamma test for table 4__________/ {
  # daylight mode
}
/_____Gamma test for table 2__________/ {
  # night mode
}
'
```

That's all for now.

frekel, 2018-01-04:
@dvv:  I am trying to get your script working, but I can not use curl:

`/home/yi-hack-v3/sbin # curl
curl: can't load library 'libz.so.1'
`

Can you provide a small howto?

Thanks,
Frank

dvv, 2018-01-04:
@frekel try putting https://github.com/utya1988/yi-hack/blob/master/sd/test/curl/libusr/libz.so.1.2.8 as /home/yi-hack-v3/lib/libz.so.1 and use
```
LD_LIBRARY_PATH=/home/yi-hack-v3/lib curl ...
```
Does it work?

frekel, 2018-01-04:
```
   ~ # cp libz.so.1.2.8 /home/yi-hack-v3/lib/libz.so.1 
   ~ # cd /home/yi-hack-v3/sbin/
   /home/yi-hack-v3/sbin # curl
   curl: try 'curl --help' for more information
   /home/yi-hack-v3/sbin # 
```
Works like a charm!

dvv, 2018-01-04:
\o/

shadow-1, 2018-01-04:
curl can also be cross-compiled without compression support. This will eliminate the dependency with libz.

josemonino, 2018-01-06:
@frekel 

> ~ # wget https://github.com/utya1988/yi-hack/raw/master/sd/test/curl/libusr/libz
.so.1.2.8
Connecting to github.com (192.30.253.113:443)
Connecting to raw.githubusercontent.com (151.101.36.133:443)
libz.so.1.2.8        100% |*******************************| 87531   0:00:00 ETA
~ # cp libz.so.1.2.8 /home/yi-hack-v3/lib/libz.so.1 
~ # cd /home/yi-hack-v3/sbin/
/home/yi-hack-v3/sbin # curl
-sh: curl: not found

could you help me?
Thanks

dvv, 2018-01-07:
@josemonino it's referenced in the script above: `# NB: curl from https://github.com/utya1988/yi-hack/tree/master/sd/test/curl`

josemonino, 2018-01-07:
@dvv Thanks a lot.

Your script is incredible. I'm looking forward to run it on my cam.
The problem is that I'm not able to get curl in my cam. What should I do?

I put https://github.com/utya1988/yi-hack/blob/master/sd/test/curl/curl into /home/yi-hack-v3/sbin/
And I put https://github.com/utya1988/yi-hack/blob/master/sd/test/curl/libusr/libz.so.1.2.8 as /home/yi-hack-v3/lib/libz.so.1 
But still not working. With that files, I get _Illegal instruction_ every time I try to execute curl.

Could you help me please?

frekel, 2018-01-07:
@josemonino:

Can you do a  ls -l from /home/yi-hack-v3/lib and /home/yi-hack-v3/sbin?

Just to be sure we have the same enviroment....

frekel, 2018-01-07:
@dvv:
The script only look to work for a couple of hours... any suggestions on how to debug this?

josemonino, 2018-01-07:
@frekel Hello Frank

```
/home/yi-hack-v3/lib # ls -l
total 406
-rwxr-xr-x    1 root     root         11052 Dec 22 19:15 libcrypt-0.9.33.2.so
-rwxr-xr-x    1 root     root          1736 Dec 22 19:16 libnsl-0.9.33.2.so
lrwxrwxrwx    1 root     root            18 Jan  1  1970 libnsl.so.0 -> libnsl-0.9.33.2.so
-rwxr-xr-x    1 root     root         38932 Dec 22 17:56 libproxychains4.so
-rwxr-xr-x    1 root     root          4348 Dec 22 19:16 libutil-0.9.33.2.so
lrwxrwxrwx    1 root     root            14 Jan  1  1970 libuv.so -> libuv.so.1.0.0
lrwxrwxrwx    1 root     root            14 Jan  1  1970 libuv.so.1 -> libuv.so.1.0.0
-rwxr-xr-x    1 root     root        125752 Dec 22 17:56 libuv.so.1.0.0
-rwxr-xr-x    1 root     root        142980 Dec 22 17:57 libwebsockets.so
-rwxrwxrwx    1 root     root         87531 Jan  7 00:48 libz.so.1
```

```
/home/yi-hack-v3/sbin # ls -l
total 882
-rwxr-xr-x    1 root     root        777293 Jan  7 01:07 curl
-rwxr-xr-x    1 root     root        125416 Dec 22 22:46 pure-ftpd
```

Thanks
Jose

dvv, 2018-01-08:
@frekel if you have SD card inserted, this is the version I've come so far to use:

```sh
#!/bin/sh

# Copyright 2018 Vladimir Dronnikov
# GPL

TELEGRAM_BOT_TOKEN=XXX
TELEGRAM_CHAT_ID=YYY
TELEGRAM_SILENT=0

curl() {
        LD_LIBRARY_PATH=/tmp/sd /tmp/sd/curl -k -q $@
}

message() {
        #curl --data "chat_id=${TELEGRAM_CHAT_ID}&text=$1" "https://api.telegram.org/bot${TELEGRAM_BOT_TOKEN}/sendMessage"
        curl "https://api.telegram.org/bot${TELEGRAM_BOT_TOKEN}/sendMessage?chat_id=${TELEGRAM_CHAT_ID}&text=$1&disable_notification=${TELEGRAM_SILENT}"
}

photo() {
        curl -F photo="@$1" "https://api.telegram.org/bot${TELEGRAM_BOT_TOKEN}/sendPhoto?chat_id=${TELEGRAM_CHAT_ID}&caption=$2&disable_notification=${TELEGRAM_SILENT}"
}

video() {
        curl -F video="@$1" "https://api.telegram.org/bot${TELEGRAM_BOT_TOKEN}/sendVideo?chat_id=${TELEGRAM_CHAT_ID}&caption=$2&disable_notification=${TELEGRAM_SILENT}"
}

ALARM=
while true; do
        test -r /tmp/sd/record/tmp.mp4.tmp && REC=1 || REC=0
        if [ "$REC" != "$ALARM" ]; then
                ALARM="$REC"
                echo $ALARM
                message "ALARM+$ALARM"
                [ "n$ALARM" == "n0" ] && rm /tmp/temp.jpg /tmp/temp.mp4
        fi
        [ -r /tmp/motion.jpg -a ! -r /tmp/temp.jpg ] && cp /tmp/motion.jpg /tmp/temp.jpg && echo JPG ready && photo /tmp/temp.jpg JPG
        [ -r /tmp/motion.mp4 -a ! -r /tmp/temp.mp4 ] && cp /tmp/motion.mp4 /tmp/temp.mp4 && echo MP4 ready && video /tmp/temp.mp4 MP4
        sleep 1
done
```

dvv, 2018-01-08:
@josemonino `Illegal instruction` -- please check if the contents of curl binary and libz.so.1 indeed the binary ones. What their sizes are?
And let us upvote having native curl in this firmware.

dvv, 2018-01-08:
@josemonino NB: curl which I've got is of 778611 octets length

frekel, 2018-01-08:
@josemonino: This is mine...  I think something went wrong with downloading curl...

```/home/yi-hack-v3/sbin # ls -l
total 883
-rwxr-xr-x    1 root     root        778611 Jan  4 19:09 curl
-rwxr-xr-x    1 root     root        125416 Dec 22 22:46 pure-ftpd
```

frekel, 2018-01-08:
@dvv: The last version of your script works like a charm.
I have edited it a small bit to hide the ALARM 0/1 and added a CAMERA_NAME variable because I have multiple camera's!

Thanks for the script!

dvv, 2018-01-08:
\o/

frekel, 2018-01-08:
Small succes!!!

[image]

I have edited the script from @dvv:
```
photo() {
        cp $1 /tmp/sd/record/last.jpg
        curl -F photo="@$1" "https://api.telegram.org/bot${TELEGRAM_BOT_TOKEN}/sendPhoto?chat_id=${TELEGRAM_CHAT
}
```
and run the following command to get a symlink:
`mkdir /home/yi-hack-v3/www/img`
`ln -s /tmp/sd/record/last.jpg /home/yi-hack-v3/www/img/last.jpg`

add the following to /home/yi-hack-v3/www/about.html (around line 82):
```
                <h2>Last picture</h2>                                                                
                <img src="img/last.jpg" alt="Last" class="img-thumbnail"> 
```

frekel, 2018-01-08:
@shadow-1:  I am trying to create a pull request, but I'm unfamiliar with the c# (?) part of your sources...

Can you point me in the right direction?
See: https://github.com/frekel/yi-hack-v3/commit/27820482d3167adeb60dfaaa9089d12bec9e99a8

frekel, 2018-01-08:
Even better:  https://github.com/shadow-1/yi-hack-v3/pull/117

shadow-1, 2018-01-09:
@frekel 
This firmware only uses (Straight) C, there is no C#. I will help you out in the pull request.

nicolasplanchon, 2018-01-25:
Hi,
Just to be sure to understand: 

I have to add your scripting code in /home/yi-hack-v3/script/alarm.sh and make it call with

if [ -f "/home/yi-hack-v3/script/alarm.sh" ]; then
        sh /home/yi-hack-v3/script/alarm.sh &
fi

in /home/yi-hack-v3/script/system.sh

dvv, 2018-01-25:
Right, that should work.

josemonino, 2018-01-25:
@dvv Thanks a lot. It's working like a charm!!

eryckm, 2018-06-05:
@frekel Have you got the images I can use with your ALARM modifications?

frekel, 2018-06-05:
@eryckm: 
Take a look here: https://github.com/shadow-1/yi-hack-v3/pull/117
I am using the last images (yi-hack-v3 0.1.5)

xhemp, 2018-06-05:
Thanks, @frekel. I asked that because of "WIP: Adding alarm functionality (and last motion in webinterface (the changes to the C file) " but looks like you were not able to get that built with the change included.

frekel, 2018-06-05:
@xhemp: Unfortunately @shadow-1 hasn't had any time to continue... Sorry!

The PR is working but you can't set settings thru the webinterface. You need to ssh into the camera and update the config-files manually (aswell ftp the new source files!)

jaxdao, 2018-09-20:
Hi @dvv 
I got this error 
~ # /tmp/sd/curl "https://api.telegram.org/bot638607717:AAEe4_Mgkh9sV7P8l7L7RFYm
i2tMhV5do4E/sendMessage?chat_id=5207785698&text=$1"
curl: (51) Cert verify failed: BADCERT_NOT_TRUSTED

Please help !

dvv, 2018-09-21:
I can not find this comment at
https://github.com/shadow-1/yi-hack-v3/issues/79 so am hoping this reply
will get the asker.
Try `/tmp/sd/curl -k https://....`.
( And never disclose your bot token! Please edit the comment source to hide
the token )

jaxdao, 2018-09-22:
HI @dvv ,
Is there a way to detect baby crying in Yi Home 1080p along with motion detection? 
As your current script, I can see ALARM 1 is motion detection and ALARM 0 is baby crying detection but has no photo and video

dvv, 2018-09-25:
@jaxdao I believe you should copy https://github.com/shadow-1/yi-hack-v3/issues/79#issuecomment-355892825 and make it watch for these files instead of /tmp/sd/record/tmp.mp4.tmp

davidgurr, 2018-12-27:
The script isn't working for me. I have no folder /tmp/sd/record.

Even if I create it, there's no files created in there while it's recording.

I'm using the latest version of the v3 hack on firmware 1.8.7.0D_201708091510

davidgurr, 2019-01-03:
@dvv do you have a timeline for exactly when the motion.jpg/.mp4/.jpg.crypt/.mp4.crypt and /tmp/sd/record/tmp.mp4.tmp files get created and deleted?

I'm having intermittent results with your script, and I think it's a timing issue.

dvv, 2019-01-14:
@davidgurr No I don't. The script is just a reactive one -- it tries to catch these exact moments.
But it just works since creation/deletion takes more than 1 second which is the period of the script.

jcconnell, 2019-05-13:
I'm trying to use the curl mentioned here in the Fritz Yi Hack. This is the error I get when trying to run that curl from the camera. Any ideas?

`./curl: line 1: syntax error: unexpected word (expecting ")")`

xhemp, 2019-05-13:
@jcconnell this project is dead, there's a yi-hack-v4 from another dev, check it out: https://github.com/TheCrypt0/yi-hack-v4

jcconnell, 2019-05-13:
I've reviewed that project but it appears he's accepting donations for RTSP functionality. RTSP is working fine for me using the project here: https://github.com/fritz-smh/yi-hack. My goal is to enable curl with HTTPS support so I can get motion notifications in Home Assistant.

xhemp, 2019-05-13:
He is, yes but it's not the only thing that is being done. RTSP is already working and also MQTT. I do have motion notifications from all my cameras on HA. I did donate, yes, the minimum is £5 so not really a bank breaker :) ,anyway, it was just a heads up.

Scope666, 2019-05-13:
I'll second what xhemp said, the Dev is crazy active and is making great progress.  The improvements to the web UI alone were worth the donation.

jcconnell, 2019-05-13:
You guys twisted my elbow. Curious if it's one donation per user or one donation per camera?

xhemp, 2019-05-13:
Per user.

jcconnell, 2019-05-13:
Done. Thanks for the heads up. Now to update 7 cameras...

[2 more replies]