shadow-1/yi-hack-v3 #81: Yi Home 27US don't pair with wifi

fmalh3iro, 2017-11-12:
Hi,

I've flash without pair the camera with the wifi, the camera start the process of paring and after read the qr code it try connect to wifi, but after a time the led goes orange and it restart the processo of parring.

I've flash with the recovery, pair with the wifi and flash the camera, the camera start try to connect to wifi (blue led) then restart (orange led). This process continues over and over.

Any ideias to resolve this problem?

shadow-1, 2017-11-13:
@fmalh3iro 
Just to confirm, you flashed the firmware from this post? https://github.com/shadow-1/yi-hack-v3/issues/40#issuecomment-315593035

What is strange is that the camera is not working even when flashed with the recovery firmware. The recovery firmware will take you back to stock firmware if flashed correctly.

When you flash the firmware, does the yellow blink for roughly 30 seconds? There are issues with some microSD cards which will prevent flashing the firmware.
Also, is your WiFi password very long?

I have found that these cameras do not work with very long WiFi passwords. The location where the WiFi password is stored does not have enough space to fit the maximum length WPA-PSK/WPA2-PSK.

xvolte, 2017-11-13:
Hello,

I was about to raise a ticket :)

The **correct hacked firmware that works** for me are located here:
home: https://www.dropbox.com/s/lxwxvm1haydkqmb/home_y18?dl=0
rootfs: https://www.dropbox.com/s/w37nkc7bz0isv2o/rootfs_y18?dl=0

The ones listed on the main page : https://github.com/shadow-1/yi-hack-v3/releases
**do NOT work** on the 27US webcam ( i have 3, all work with the dropbox link, but none in the github release folder).

Could you update them @Shadow-1 ?
i got them from : https://github.com/fritz-smh/yi-hack/issues/174

Best regards!

fmalh3iro, 2017-11-13:
Thx xvolte for the files, I've used the files listed on the main page.
Best regards,

closed 2017-11-13 by fmalh3iro