shadow-1/yi-hack-v3 #82: Is there any plan for Chinese version of YI Outdoor Camera?

HuipengRen, 2017-11-20:


HuipengRen, 2017-11-20:
Or is there any way to do it by myself based on this project? I bought one two weeks ago, but I just found that I can not use it in the US unfortunately.

shadow-1, 2017-11-20:
@HuipengRen 
This is a very interesting camera which I would like to support. However as I don't have this camera, I don't think I will be able to develop custom firmware for it yet.

I can guide you through a few basic steps to confirm whether the Yi Outdoor camera is similar to the other cameras I currently support. If it is very similar to the Yi Home or Yi Dome cameras, I should be able to support it without too many issues, even if I don't have the camera.

Unfortunately the only way to confirm this is to get access to the cameras serial interface which requires the camera to be opened up along with some soldering. If you are willing to do this, I may be able to help.

javier26, 2017-12-05:
I am willing to help. I have the "YI Outdoor Security Camera, Cloud Cam Wireless IP" as described on amazon and sold to US customers.

shadow-1, 2017-12-06:
@javier26
That is great. The first thing that you need to do is get access to the console of the camera. All the Xiaomi cameras I have seen have a serial port connection on the main board.

Older versions of the camera had a connector whilst newer ones only have pads which can be soldered.

Once you have the serial port connection soldered, you will need to use a USB-Serial converter to access the console.

You can use this YouTube video as a go by. It is for the original Yi Home and not the Yi Outdoor 1080p. Hopefully it isn't much different.
https://www.youtube.com/watch?v=3e8jXuxiRzY

When you manage to open your camera to get access to the main board. Can you post some high resolution photos of the internals here? I'm sure it will be helpful to others.

javier26, 2017-12-06:
I have no idea how to open this thing. There are no visible screws apart from the ones that hold the microSD slot door closed.  [https://photos.google.com/photo/AF1QipPDgmiWniilMRcZFrTphh8bnQ9Ssb4V446K_Y71](url)
[https://photos.google.com/photo/AF1QipO70dAKY7yTYoXiBAkJd6krVm6umKOBtedLiXPW](url)
[https://photos.google.com/photo/AF1QipMVavnD5VDBBLimDtA2-dbLa8Rl54_Jay-wc93g](url)

I tried to pry open from the front with a little pressure but it wouldnt give. Ill keep trying

HuipengRen, 2017-12-07:
I actually unlocked the region lock without opening the camera couple weeks ago.

javier26, 2017-12-07:
Are you able to enable rtsp on it and remove the cloud recording, and all the good stuff?

HuipengRen, 2017-12-07:
No, it only unlocked the region lock, so basically
1. Set up a dns server with dnsmasq in your laptop, and config  "api.xiaoyi.com" mapping to your laptop ip.
2. Log in to your wifi router and change the dns server to your new dns server (your laptop ip).
3. Set up a customize transparent http proxy on your laptop, which will forward to another http proxy in the Mainland China. I can not config squid or other http proxy to do it correctly, so I just wrote a very simple one. 
4. You will need to set up the above again when the camera restarts.

HuipengRen, 2017-12-07:
To unlock the region lock, I think you can also simply do ssh port forwarding for port 80/443 to a instance in a cloud provider in China (like aliyun or mtyun) after redirecting your dns server.

HuipengRen, 2017-12-07:
I can share the simple transparent http proxy if anyone needs it.

shadow-1, 2017-12-07:
@HuipengRen
This is almost the same method my custom firmware uses to bypass the region lock. The main difference of course is that the firmware implements everything on the camera. No need for special routers or converting a spare computer into a transparent proxy server.

@javier26
Unfortunetly to develop a custom firmware, we need to find a way to open the camera. I guess it is much more difficult to open because it is dustproof and waterproof.

chenkok, 2017-12-18:
So, basically if you have physical access to the Yi Outdoor cam then you'll be able to make a custom firmware right? Then we'll be able to just flash the cam using a memory card? 
I bought a few awhile ago for my dad without knowing it has region lock.  So, i might be able to give you one for testing. Is that ok with you?

shadow-1, 2017-12-18:
@chenkok 
With physical access to the camera, we can do investigations within the embedded operating system and determine what needs to be done to hack the firmware.

This particular custom firmware (yi-hack-v3) is specific to Xiaomi cameras based on the HiSilicon Hi3518e chipset. If the Yi Outdoor is also based on a HiSilicon chipset, there is a good chance that this firmware can be adapted to provide Yi Outdoor 1080p support without a lot of modification. However if the Yi Outdoor camera is based on a different chipset, a completely different hack will need to be developed, potentially with a different way of applying firmware updates (if at all possible) and potentially a different method of hacking the firmware.

You never know with Xiaomi. For example, the Yi Home 2 camera, despite the similar product name, similar look and feel, it is a completely different camera to the original Yi Home series of cameras. The Yi Home is based on the HiSilicon Hi3518e chip whilst the Yi Home 2 is based on the Ambarella S2LM chipset. A completely different firmware was developed to hack the Yi Home 2 camera because it is based on a different chip.

I am happy to have a look at the Yi Outdoor for testing purposes. However only after investigation and testing can we find out what needs to be done to hack the firmware. There is also a chance that it cannot be done easily.

chenkok, 2017-12-19:
@shadow-1 Can you provide your address, will go get a quote from courier service.

ykhandler, 2017-12-20:
Interesting...

shadow-1, 2017-12-25:
@chenkok 
Can you provide me with an email address or a way I can send you my address privately?

However you can get a quote from your courier company to send to Melbourne, Australia. It should be exactly the same cost to send a parcel anywhere within Melbourne.

earth2004, 2017-12-29:
@shadow-1 Just received my Chinese version yi outdoor camera today and couldn't have it work because the region lock and I came across to this page. I live in Melbourne and am willing to send you one for testing. Kindly email me liuwenzheng2004@gmail.com and thank you in advance!

earth2004, 2017-12-29:
@HuipengRen Can you share the transparent http proxy to liuwenzheng2004@gmail.com please?

mugennam, 2017-12-30:
@HuipengRen may i get a copy of the trans proxy too qwer@telus.net thanks

HuipengRen, 2017-12-30:
@earth2004 @mugennam I just put it here, https://github.com/HuipengRen/yihttptunnel

shadow-1, 2018-01-04:
@earth2004
I just sent you an email. Let me know if you are willing to send me a Yi Outdoor 1080p for development purposes.

milanzelenka, 2018-01-15:
I would like to contribute to discovery YI Outdoor via serial, but it looks like impossible to open (dismout) camera without loosing it's watter resistance... :-(

There are no screw (only for sdcard cover), no hidden screw (under back-side label). I tried to unfold the join on perimeter in front, but unsuccess. If you have any idea how to dismout, please share it...

These YI cameras without this custom (hacked) firmware is just a piece of toy...

jasperpants, 2018-01-17:
I'd like to contribute as well.  I took a look at the camera when I received it and I think you'd basically have to destroy the case to get in to it.  Maybe if there is enough of us chipping in $10 each we can get a sacrificial one to shadow-1.

ultimacarlos, 2018-01-22:
I bought the cam without knowing the region block. Have been searching the net on how to open the case and this is the best info I found thus far. [https://fccid.io/2AFIB-YHS3017/Internal-Photos/Int-Photos-3558245](url). Hope this will help to develop the hack for it. Thanks.

JustinTrouble, 2018-01-22:
Looking at those photos, it's obvious that the black cover first needs to be removed to take apart the camera.
It looks like the black cover is simply glued in place. You could try heating the edge of the black cover with a heat gun or hair dryer and try to pry and lift it off.

ultimacarlos, 2018-01-22:
@JustinTrouble 
You're right. The black cover is glued on the circumference. Can ply off easily by popping a thin object on the top hole. Be cautious not to push all the way in as there's a film behind the small hole for the light sensor, I guess. Just stick it in enough for plying. Next, you will need a slim and long screw driver to take off the 4 screws that hold the whole assembly.

milanzelenka, 2018-01-22:
I successfully opened camera (thanks to @ultimacarlos for link). I used hair dryer and small screwdriver (in the sensor hole) to release front glass...

TTL connected, serial established on speed 115200 8N1. I've captured boot log of original firmware. here: https://pastebin.com/C9AcUJHi

What can I try? Is there any interactive shell to gathering some diagnostic data?

milanzelenka, 2018-01-22:
Another bootlog with camera powered from original USB cable and adapter (not from UART): https://pastebin.com/DepPb3D8

At the end of bootlog camera was successfully connected to wifi.

jasperpants, 2018-01-24:
@milanzelenka were you able to interrupt uboot at all?  The youtube video @shadow-1 linked to on Dec-6th starts getting interesting around the 13 minute mark where he changes the boot args and eventually gets root on the camera.

milanzelenka, 2018-01-25:
@jasperpants Ctrl-C (or any other key/s) does not work :-( It's look like there is bootloader timeout set to 0 sec and it's impossible to break it. Kernel starts boot as soon as UBOOT message was shown.

```
### UBOOT:test pin high! ###
Hit any key to stop autoboot:  0
```

What could be interesting is mentioned "test pin". I don't know what it is, maybe some pin on main board which is needed to connect to GND or VCC? I don't see any pin labeled with TEST :-( I'll keep trying...

tiong0526, 2018-01-26:
Does any experts here managed to develop a custom firmware for outdoor camera?? I bought 2 units and didnt know it locked for mainland china used only. Many thanks in advance.

HuipengRen, 2018-01-27:
54.84.30.91
This is my own DNS server (and transparent HTTP proxy), which is working well for my camera. If you want to have a try you can simply specify the dns server with this IP on your wifi router.
Note, I can not guarantee any reliability since it's just a single instance on Amazon EC2.

milanzelenka, 2018-01-27:
I've just littlebit investigated about Yi outdoor camera and custom firware. The Problem is mainly in the uboot bootloader timeout set to 0. The only way how to change it is rewrite it directly into env_y30 image and flash it via sdcard. BUT where to get original env_y30 image? 

1. Directly from flash via some usb programmer, but I don't have any experience with it. Do you?

2. From original Yi firmware image downloaded from yi support page. [Here](https://github.com/fritz-smh/yi-hack/issues/124#issuecomment-275670843) is very well described procedure from @shadow-1 to unpack/decode original firmware to get partition images, but in step 4. we need to more investigate how rsa_pub_dec script works.

3. Build env_y30 image from scratch and flash it. Without backup. Yes, it's crazy, it's fast way to hell...

Do you have any idea? Any experience with read flash directly via usb programmer?

ultimacarlos, 2018-01-28:
@milanzelenka would the 2nd method in this link help? http://processors.wiki.ti.com/index.php/Change_U-Boot_bootdelay_setting

shadow-1, 2018-01-28:
@milanzelenka 
I have been away for a little while and I am starting to catch up. Based on your log files, the Yi Outdoor camera is indeed almost identical to the other cameras I currently support.

I should be able to create a version of yi-hack-v3 which is compatible with the Yi Outdoor with a bit help from you as you have access to the camera via serial port.

---

Can you post the boot log with a microSD card inserted (it might need to be formatted to FAT32). The boot log should tell me what the filenames need to be for the custom firmware files to flash successfully. This information only appears if the camera is booted with a formatted microSD card inserted.

---

Can you post the output from the following command:
```
cat /proc/cpuinfo
```

This should confirm exactly what processor is running on the camera. This will tell me which compiler I should use when compiling custom software for the camera.

---

The most important thing you can do is backup the partitions on your camera. This will allow me to create Recovery images so that it will be possible to go back to stock firmware after experimenting with the camera. In addition, it forms the basis for my custom firmware images.

I start with an unmodified image, then alter it to add the additional features and programs. This allows me to create a firmware image which extends the functionality of the camera whilst still allowing the official Xiaomi smartphone app to work.

To backup the cameras partitions to the microSD card, execute the following commands:
```
dd if=/dev/mtd0 of=/tmp/sd/mtd0
dd if=/dev/mtd1 of=/tmp/sd/mtd1
dd if=/dev/mtd2 of=/tmp/sd/mtd2
dd if=/dev/mtd3 of=/tmp/sd/mtd3
dd if=/dev/mtd4 of=/tmp/sd/mtd4
dd if=/dev/mtd5 of=/tmp/sd/mtd5
dd if=/dev/mtd6 of=/tmp/sd/mtd6
dd if=/dev/mtd7 of=/tmp/sd/mtd7
```

This assumes that the microSD card is mounted to `/tmp/sd` like the other cameras I support.

When you post the partition files (mtd0 to mtd7), I can start work with creating a version of yi-hack-v3 which is compatible with the Yi Outdoor camera.

milanzelenka, 2018-01-28:
@shadow-1 Great! Thanks...

Here is boot log with clean SD card inserted: https://pastebin.com/T1TZ1vpp

The problem is there are no way to enter commands. Shell is not executed at the end of system start. I can't break uboot to change kernel params (because of timeout 0). I tried Ctrl-C in combination with reset button, etc..

shadow-1, 2018-01-28:
@milanzelenka 
For all the other Yi cameras I played around with, there is no need to login to the camera to access shell. All I ever needed to do is press Enter a few times to see the command line interface prompt. However the Xiaomi programs would constantly write to stdout which makes it very difficult to type commands or read the output from commands.

I had a quick look at your log file and it appears you have already tried pressing Enter a few times, pressing Ctrl-C etc and nothing happened. We might have to backup the partitions on the camera via startup script rather than via shell.

milanzelenka, 2018-01-28:
@shadow-1 yes, It seems that shell is really not started after init... :-( The same behavior if camera is connected to wifi and after factory reset.

RX cable seems to work, because if I disconnect RX, I can't write (I don't see anything writen on console). 

You mentioned some startup scripts? What is it?

shadow-1, 2018-01-29:
@milanzelenka
What I will do is examine the firmware and see if the camera executes a script from the microSD card.

What I have found is that most Xiaomi cameras execute a script from the microSD card upon bootup (when on stock firmware). However quite often this functionality is disabled during future firmware updates.

If this is the case for the Yi Outdoor, we may be able to back up the partitions without access to the console via serial port.

Your bootup log confirms the filenames required to flash the firmware.

milanzelenka, 2018-01-29:
@shadow-1 
I had only old one sdcard (2GB) and camera didn't mount it:
`checkdisk.c(main-387)[00:00:03.260]:ERROR! TF card space is too small: 1876 MB`

Today I bought new 16GB microSD card and now SD is mounted:
```
checkdisk.c(do_fat_check-140)[00:00:04.259]:Mount sd ok.
checkdisk.c(check_partion_vfat-156)[00:00:04.259]:TF Card check finished.
```
I tried to make some shell script and copy it to sdcard as /factory_test.sh, /test/factory_test.sh, equip_test.sh, /test/equip_test.sh, etc... but nothing happened. :-(

According to log:

`/home/app/init.sh: line 10: /home/app/script/factory_test.sh: not found`

Seems like that /home/app/script/factory_test.sh which is calling factory_test.sh on SDcard is deleted.

When I press the button under SDcard slot, console shows this message:

`[CPLD_PERIPH] factory test!`

...but nothing will happen.

Is any other possibilities to backup partitions? Thanks...

milanzelenka, 2018-02-05:
1. Maybe we would create completly new HOME partition (from scratch or from another model of camera) which has modified /home/app/init.sh to start shell.

2. Then we would make backup of other partitions (exept /home).

3. With this backup, it would be easy to make modified ENV partition to change u-boot timeout.

4. Someone else with same camera flash this new ENV partition and make backup of HOME.

Result: We have complete backup of this camera model for future development...

@shadow-1 what do you think about this procedure? Are you willing to help me create home partition?

mcafici, 2018-02-23:
Hi, I have an "Yi outdoor" , and I'm very interested in your work to enable  the RSTP server. You have some news.

chenkok, 2018-03-04:
@shadow-1 hi, my addy is chenkok.chong@gmail.com

RaulSe89, 2018-04-11:
I have 3 Yi outdoor cams and would like to use them with RSTP. I am also very interested in this update.

steven-shi, 2018-04-12:
@neversleepnight at the moment, I can only using proxy to make it running but cannot flash the custom firmware

mikfaina, 2018-05-07:
@milanzelenka @shadow-1 any news?

dylangerdaly, 2018-05-07:
Well, I'm late to the party! Being board I've ordered 2 of these devices, let's get RTSP on this thing.

It looks like in the FCC Picture there's a 8 Pin SOIC Flash Chip, looks like it could be Windbond? I'll hook this badboy up to my RPi and see if I can dump the entire flash, perhaps we can append `init=/bin/sh` to the CMDLINE and get a shell over UART.

Let's get into the firmware and see if `[CPLD_PERIPH] factory test!` is doing anything silly (Like running scripts on the SD Card)

Anyone have a link to the firmware? Would save me REing the Yi APK.

dylangerdaly, 2018-05-07:
Gah, they've encrypted the firmware, they really don't people messing around in here
```
00000000  33 2e 30 2e 30 2e 30 42  5f 32 30 31 37 31 32 31  |3.0.0.0B_2017121|
00000010  31 31 39 33 36 0a 73 aa  1f 9a fe ea 17 76 78 d6  |11936.s......vx.|
00000020  ac a8 f1 a5 61 18 8f 66  b8 3b 9a 57 c6 ce cd c0  |....a..f.;.W....|
00000030  e0 bb 4c 9f 39 64 32 eb  e2 c1 b1 72 12 8f e1 32  |..L.9d2....r...2|
...
```

No useful strings, binwalk comes up empty.

dylangerdaly, 2018-05-07:
According to the APK, it connects to the Camera on 192.168.0.1:4444

```java
...
        private void b() {
            AntsLog.d("UpdateFirmwareTcpService", "ThreadTcpUpload initialize");
            try {
                this.d = new Socket();
                this.d.connect(new InetSocketAddress("192.168.0.1", 4444), 3000);
                this.e = new DataOutputStream(this.d.getOutputStream());
                this.f = new FileInputStream(new File(UpdateFirmwareTcpService.e(this.a).b()));
                this.c = new byte[0x400];
                while(true) {
                    int v0_1 = this.f.read(this.c, 0, this.c.length);
                    this.b = v0_1;
                    if(v0_1 > 0 && !this.i) {
                        this.e.write(this.c, 0, this.b);
                        this.e.flush();
                        this.g += ((long)this.b);
                        this.h = ((int)(this.g * 70 / UpdateFirmwareTcpService.e(this.a).a()));
                        Message v0_2 = UpdateFirmwareTcpService.f(this.a).obtainMessage();
                        v0_2.what = 5;
                        v0_2.arg1 = this.h;
                        UpdateFirmwareTcpService.f(this.a).sendMessage(v0_2);
                        continue;
                    }

                    break;
                }
...
```

dylangerdaly, 2018-05-07:
If it _is_ listening on 192.168.0.1:3333 you should be able to start POSTing JSON stuff at it.

dylangerdaly, 2018-05-08:
So be super careful with the VCC PIN on the MX25L12835F, I just bricked one of mine :(

batrarobin, 2018-05-08:
There might be some useful info in this thread with regard to extracting encrypted firmware image: https://github.com/shadow-1/yi-hack-v3/issues/23

dylangerdaly, 2018-05-08:
It looks like he's extracting a jffs2 image, however the firmware for this device appears to be encrypted, I'm waiting on my gear to arrive so I can extract the firmware directly from the SPI Flash (MX25L12835F)

Once there, we can trawl thru binaries and see what they're doing, maybe executing something on a flash card.

dylangerdaly, 2018-05-08:
I can confirm it's 100% encrypted firmware

```
...
1+0 records in
1+0 records out
22 bytes (22B) copied, 0.001782 seconds, 12.1KB/s
Hisilicon Media Memory Zone Manager
Load hi_cipher.ko success.
1+0 records in
1+0 records out
22 bytes (22B) copied, 0.000359 seconds, 59.8KB/s
83017+1 records in
83017+1 records out
1826384 bytes (1.7MB) copied, 1.761122 seconds, 1012.8KB/s
1+0 records in
1+0 records out
1344 bytes (1.3KB) copied, 0.000397 seconds, 3.2MB/s
1357+1 records in
1357+1 records out
1825040 bytes (1.7MB) copied, 0.058107 seconds, 30.0MB/s
encrypt file is enc_key, and save decrypt file to dec_enc_key

prepare: open read and write files
crypt start
update
27666+1 records in
27666+1 records out
1825998 bytes (1.7MB) copied, 0.577843 seconds, 3.0MB/s
1+0 records in
1+0 records out
33 bytes (33B) copied, 0.000348 seconds, 92.6KB/s
1+0 records in
1+0 records out
33 bytes (33B) copied, 0.000359 seconds, 89.8KB/s
1+0 records in
1+0 records out
22 bytes (22B) copied, 0.000348 seconds, 61.7KB/s
is not
is not
update new file home/homever
/tmp/update
...
```

dylangerdaly, 2018-05-12:
FYI, this is uboot code for the specific chip set running on the camera, hypothetically, we should be able to compile this and replace it on the camera.

https://github.com/49handyman/u-boot-hi3518ev200

dylangerdaly, 2018-05-12:
## Update - It works!

It works!

Using an RPi, I was able to replace the old jffs2 file system with @shadow-1's hacked version and it appears to be working correctly.

```
root@192.168.35.70's password: 
Welcome to HiLinux.
~ # id
-sh: id: not found
~ # echo lol
lol
~ # ls
~ # cd /ls
-sh: cd: can't cd to /ls
~ # cd /
/ # ls
bin      dev      etc      home     lib      linuxrc  mnt      proc     root     sbin     sdcard   sys      tmp      usr      var
/ # 
```

Even more interestingly, it remembered my NV Data, so it connected to my WiFi and even shows up in the App.

Looks like the camera itself works perfectly, I'll update in the coming days!

It looks like the Yi 1080p Home is using the exact same SoC, meaning all of the pinout for the camera, leds etc are the exact same. This is why its working.

fredyagu, 2018-05-12:
Hi, @dylangerdaly , what is RPi? Can you make a tutorial how to do it?

dylangerdaly, 2018-05-12:
@shadow-1 did you remove RTSP Server? The cloud and cloud-api binaries look so much like backdoored malware my goal is to turn this device into a dumb RTSP Stramer Cam

dylangerdaly, 2018-05-12:
@fredyagu you'll need a RPi3 and a SOIC8 clip, then you'll need to tear the device down, trying to see if we can maybe mess around with the updater.

But I'm 2 Piña Colada's in, tomorrow I'll see if it's possible to mess with the updater (Create an OTA that dosen't care about encryption / isn't encrypted)

@shadow-1 seems to be software related, creating my own jffs2 images is successfull

After resoldering
```
[+] Launching Dropbear...
dropbear: can't load library 'libutil.so.0'
```

Absolutly Sucks...

At least I had it execute off the SD-Card

```bash
echo "[+] Launching Dropbear..."
dropbear -R

# If Dropbear Dosen't Work, have a 2nd way in
if [ -f /tmp/sd/fuck_sakes.sh ]; then
	echo "[!] Fuck Sakes Detected... Running Script on SD-Card..."
	export LD_LIBRARY_PATH=/tmp/sd/lib:$LD_LIBRARY_PATH
	export PATH=/tmp/sd/bin:$PATH
	/tmp/sd/fuck_sakes.sh
fi
```

dylangerdaly, 2018-05-12:
Lol okay, I'm an idiot

```
System startup
=========1
hello, open console

 do_auto_sd_update: detect flash size: 0x1000000, gsdexist=1 
Interface:  MMC
  Device 0: Vendor: Man 275048 Snr 01ce14b1 Rev: 3.0 Prod: SD32G
            Type: Removable Hard Disk
            Capacity: 29800.0 MB = 29.1 GB (61030400 x 512)
Partition 1: Filesystem: FAT32 "NO NAME    "
      190   fuck_sakes.sh 
            lib/
            record/
            bin/
   962448   rootfs_h30 
    41472   log_first_login.tar.gz 
    36864   log_wifi_connected.tar.gz 
  7696000   home_h30 

5 file(s), 3 dir(s)

1: fuck_sakes.sh!
2: rootfs_h30!
 find index 4: rootfs_h30 

3: log_first_login.tar.gz!
4: log_wifi_connected.tar.gz!
5: home_h30!
 find index 5: home_h30 

u-boot_h30 not found!

env_h30 not found!

conf_h30 not found!

kernel_h30 not found!
reading rootfs_h30

magic 0x27051956 0x27051956 
arch 0x2 0x2 
size 0xeaf50 0x40 
type 0x7 0x2 
crc 0xffe70a37 
name 0.1.5-hi3518-rootfs 
idx4: old : 59a7a584, new : 5a3f82a0
reading rootfs_h30
update finished, calc crc star...
Erase env area success!
reading home_h30

magic 0x27051956 0x27051956 
arch 0x2 0x2 
size 0x756e40 0x40 
type 0x7 0x2 
crc 0xbb15f81f 
name 0.1.5-hi3518-home 
idx5: old : 59e02d01, new : 5a3f82af
reading home_h30
```

It's possible to load @shadow-1's images just by renaming `home_y20` to `home_h30` and `rootfs_y20` to `rootfs_h30`

[74 more replies]