shadow-1/yi-hack-v3 #83: [ideas] Any thoughts on disabling rotation on Yi Dome

airtien, 2017-11-21:
Sometimes the Yi Dome simply overrotate can facing the wall instead of the room. I would be very gratful if the Yi Dome can be "fixed" and stop auto rotation through settings. I understand I should have bought Yi Home instead of Yi Dome. But is there any "fixes" that can be applied to the firmware side to disable the rotation feature..

Many thanks to the great work.

shadow-1, 2017-11-22:
@airtien 
This is a little unusual. I have not had issues with the camera moving towards the wall by itself.

Can you confirm if Cruise Mode and Motion Tracking is disabled in the Yi Home smartphone app.

dogger23, 2018-08-15:
I have confirmed this issue as well. It seems to rotate regularly, between every 15 to 60 minutes. It occurs for me when it is on cruised bookmarked location, with either 2 or 3 set locations. I already had one camera RMAed the second had the same issue. This happens on both with stock or yi-hack-v3.