shadow-1/yi-hack-v3 #84: firmware files compromise the camera bootup

webberd1wei, 2017-11-27:
I have a Yi home camera and I loaded the firmware files rooffs_18 and home_18 to the sd card. The camera would boot to solid blue LED, but after 10s it would reboot automatically and the cycle goes on.

Any idea what went wrong and whether I can recover from it？( The cycle still exists even after I remove the firmware files)

ElRojo, 2017-11-27:
Hey @webberd1wei 

I was able to reboot mine by putting the following files in the root of the SD card. I've zipped them for you, so just unzip and place those at the root location!
 
To be honest, I'm not sure if they all need to be there -- but it worked for me! :) 

[Recovery Package. US27.zip](https://github.com/shadow-1/yi-hack-v3/files/1504965/Recovery.Package.US27.zip)

webberd1wei, 2017-11-27:
Hey @ElRojo It works for me! Thank you for saving my camera!

closed 2017-11-27 by webberd1wei