shadow-1/yi-hack-v3 #86: YI Outdoor Camera 1080p

rish197, 2017-11-29:
Hi - Any plans on adding custom firmware for the new YI Outdoor Camera 1080p? Firmware 3.0.0.0A_201710131102? I'm looking to enable RTSP.

shadow-1, 2017-11-30:
@rish197 
I would like to develop a firmware for this camera as I find a proper outdoor rated camera quite interesting. However as I don't have this camera, I don't think I will be able to develop custom firmware for it yet.

I can guide you through a few basic steps to confirm whether the Yi Outdoor camera is similar to the other cameras I currently support. If it is very similar to the Yi Home or Yi Dome cameras, I should be able to support it without too many issues, even if I don't have the camera.

Unfortunately the only way to confirm this is to get access to the cameras serial interface which requires the camera to be opened up along with some soldering. If you are willing to do this, I may be able to help.

stonerl, 2017-12-07:
@shadow-1 where do you life? Maybe I could donate one...

shadow-1, 2017-12-08:
@stonerl
I currently live in Australia.

danxfisher, 2017-12-25:
has there been any update on this one by chance?

shadow-1, 2017-12-25:
@danxfisher 
I don't have this particular camera, so I have not been able to develop a custom firmware for it.
However if you are willing to open up your camera and solder leads to the serial port connection on the main board (all Xiaomi cameras I have seen have a serial port connection), you may be able to help.

There is some further information available in this thread: https://github.com/shadow-1/yi-hack-v3/issues/82

aocana, 2018-03-22:
any update related to this thread..?

dvbit, 2018-04-24:
I would gladly contribute to buy a camera does anybody wants to?

dylangerdaly, 2018-05-07:
Any update? I'm looking to buy a few of these devices, I'm willing to get root on the devices. How easy would it be to enable RTSP?

dylangerdaly, 2018-05-15:
Updates here --> https://github.com/shadow-1/yi-hack-v3/issues/82

jamieshaw, 2018-07-27:
Has there been any movement on this, the trail appears to have gone cold… 😢

dawsmac, 2018-08-15:
Anyone had any joy with this camera, I have one and happy to test firmware ? 

Does it need to be opened up etc??? 

Cheers

Rich

Lecl0ne, 2018-08-23:
Very interested in it too, I have the camera.

Willytoons, 2018-10-06:
Estimados amigos, acabo de recibir mi camara outdoor YI comprada por amazon, y NO PUEDO configurarla, tengo que decir que ya tengo hace medio año la YI HOME, 
-Lo primero que llamó mi atención es que la camara me emite sus status en idioma CHINO-
-Sigo todas las indicaciones y aparentemente se llega a conectar a mi wifi, pero al final me dice hay un error en la selección del disposito (es imposible porque lo ubiqué con el mismo QR).
-Estoy tratando de actualizar el firmware, pero a pesar que lo he descargado y colocado en una Micro SD, se lo he puesto a la cámara, pero parece no ocurrir nada, leí por ahí que hay que renombrar el archivo del firmware.
Tuvieran la gentileza de ayudarme e indicarme que puedo estar haciendo mal, ya he reseteado la camara casi 20 veces y no logro configurarla.
Gracias por su tiempo.
PD (con la otra camara YI HOME nunca tuve problemas

Willytoons, 2018-10-06:
@aelg305 
estimado, gracias por tu respuesta , pero he llegado a este enlace que publicas en el post que mecionas, pero no hay nada
https://github.com/shadow-1/yi-hack-v3/issues/228

aelg305, 2018-10-06:
@Willytoons 
ya hiciste esto?

esta en el post #228 es uno de mis comentarios...literalmente es el "paso por paso"
-formatted the card
-downloaded the files from yi-hack (roof_y20) and renamed them to h30 and included my wpa_supplicant.conf and let it boot (yellow light comes one, then flashing yellow (meaning is flashing) and then solid yellow but no connection yet
-disconnected and drop the files from fedyfausto with the h30 and y30 and connected again
-yellow -> flashing -> solid -> connected after a couple of turn on/off from power
-telnet into it to confirm -> OK
-copy the following 4 files to the home directory using telnet (rtsp2302, rtsp2302.sh, libstdc++.so.6, init.sh)...used command: cp *.sh .6 rtsp /home/app
-sealed the deal with a power cycle (disconnect from wall/reconnect)

mikfaina, 2018-10-08:
English please

arunsmu, 2018-11-11:
> @Willytoons
> ya hiciste esto?
>
> esta en el post #228 es uno de mis comentarios...literalmente es el "paso por paso"
> -formatted the card
> -downloaded the files from yi-hack (roof_y20) and renamed them [...]

Can you please make a youtube video showing how its done. It'll be extremely helpful

badziomiki, 2018-12-07:
Can anyone tell me which exactly image I should use? Also, does this hack enable ftp? i need that for homeassistant integration