shadow-1/yi-hack-v3 #87: Yi Home Camera 720p 47US

floari, 2017-11-30:
Hi,

do you plan on supporting the 47US (Yi Home Camera 720p) Version? I think its the european version of the camera.

Thanks,

shadow-1, 2017-11-30:
@floari
I was not aware that Xiaomi released a new version of their 720p camera (47US). 

As I don't have this camera. I will need your help to support this camera.

By any chance, are you able to get serial port access on the camera? This involves some soldering on the cameras main board.

If you are willing to solder fly leads to the serial port on the camera and use a USB-RS232 converter to gain access to the cameras console. I should be able to guide you in obtaining a backup of all the partitions on the camera (which I will need to develop a firmware) and guide you through some checks that I will need to do on the camera.

floari, 2017-11-30:
@shadow-1 
Sorry - I dont have any of the tools you mention. Is it possible to try the 27US Version of the Firmware for the device (and not to brick it)?

shadow-1, 2017-11-30:
@floari
You can try flashing the firmware for the 27US version. If the expected firmware file is different to the 27US, nothing will happen. The firmware upgrade will be skipped. However if the expected firmware file is the same, the camera will attempt to flash the firmware.

Unfortunetly there is no guarantee that the 27US firmware won't brick the camera. As I don't have a back up of the partitions on the 47US version of camera. I won't be able to create a recovery image to go back to stock firmware.

koncode, 2017-12-08:
I have the 47US camera. I tried to flash the 27US version and it upgrades smoothly but unfortunately it does not work afterwards. It dumps, however some log files on the SD card(log_no_wifi_dev.tar.gz). 

After that i tried to restore using the 27US recovery file and it restored my camera without any problems. 
I can repeat the process and provide these log files if you think they would be useful.

shadow-1, 2017-12-09:
@koncode 
This is very interesting! It sounds like the Yi Home 47US is almost identical to the Yi Home 27US.

Can you try using the pre-release firmware from this post and see if it works: https://github.com/shadow-1/yi-hack-v3/issues/40#issuecomment-315593035
If it works, can you let us know.

koncode, 2017-12-09:
Yeah, it works with the pre-release firmware!

WiFi, ftp, HTTP and telnet fully operational. Plus it connects with the app as expected.
Only the RTSP feed is not working but is see this is a known issue for the 27US version as well.

ordimans, 2018-01-18:
@shadow-1 
I have 47US, if i can help you about partitions.
I have use a USB-RS232 converter, can you show me where is pin on the board ? i will solder on it.
After, just need to connect with putty ?