shadow-1/yi-hack-v3 #89: How to take a pic automatically ?

nicolasplanchon, 2017-12-04:
Hi there,
This hack is simply perfect so far!

I'm working on a student project and I have just a little matter:

When the cam has detected a motion,do you know if it's possible to take a pic rather than a movie please ? :)

Best regards,
Nicolas.

shadow-1, 2017-12-04:
@nicolasplanchon 
Have a look at this post and see if this helps you a bit: https://github.com/shadow-1/yi-hack-v3/issues/79#issuecomment-348161594

I have not verified this. However it appears that a JPEG file is created when the camera detects motion for a few seconds. I suspect this image is the thumbnail that you see in the Yi Home app for motion alerts.

A program or script could be developed to check if this JPEG file exists. Then an action could be taken if it is found (I'm not exactly sure what you would like to achieve). Ideally this would be event driven, a continuous loop that constantly checks if a JPEG file exists would not be efficient.

nicolasplanchon, 2017-12-04:
To check and extract automatically I have already found this script: https://github.com/AndMu/YiScanner

I just need to know how to capture an image rather than a video :p

JoaquinBeceiro, 2019-05-19:
For install curl:

`cd /home/yi-hack-v3/sbin/`
`wget https://github.com/utya1988/yi-hack/raw/master/sd/test/curl/curl`
`chmod +x /home/yi-hack-v3/sbin/curl`
`cd /home/yi-hack-v3/lib`
`wget https://github.com/utya1988/yi-hack/raw/master/sd/test/curl/libusr/libz.so.1.2.8 -O libz.so.1 `