shadow-1/yi-hack-v3 #90: YI HOME 1080P

JohnMaxwel, 2017-12-04:
I have a problem with Yi Home 1080P the camera have no functions. When I plug it then I have only flashed yellow light. I can not connect to FTP or anymore, as current files are damaged in my opinion.

I will downgrade to Official Firmware without yi-hack is this possible? 
So I want to replace the software with the original version. I hope that it is possible to get the original software via MicroSD.

shadow-1, 2017-12-04:
@JohnMaxwel
Yes it is possible to go back to default firmware. I have provided instructions on the main page. Have a look at the following section on the main page: https://github.com/shadow-1/yi-hack-v3#going-back-to-stock-firmware

I always want to provide an option to opt-back to the default/stock firmware. A lot of other firmwares do not provide such an option and sometimes unexpected things happen when the firmware modifies files on the filesystem without providing a way to undo the changes.

JohnMaxwel, 2017-12-06:
@shadow-1 
Thank you!! It worked. 

I thought I would not be working anymore but I succeeded!

shadow-1, 2017-12-06:
@JohnMaxwel
That is great news!

closed 2017-12-06 by shadow-1