shadow-1/yi-hack-v3 #91: Yi Home 27US Camera unable to connect to wifi

feiming, 2017-12-05:
I follow the instruction https://github.com/shadow-1/yi-hack-v3 till step 9, where it's connecting. Then it won't connect and restart. I'm using yi-hack-v3 0.1.4-beta2. 

I hope this [log.txt](https://github.com/shadow-1/yi-hack-v3/files/1531833/log.txt) taken from log_no_wifi_dev.tar.gz will help

filo4u, 2017-12-05:
Yes 27US from posted links don't work. Try these links from fritz-smh : https://github.com/fritz-smh/yi-hack.
This is the 0.1.4-beta2
https://www.dropbox.com/s/lxwxvm1haydkqmb/home_y18?dl=0
https://www.dropbox.com/s/w37nkc7bz0isv2o/rootfs_y18?dl=0

shadow-1, 2017-12-06:
@feiming
@filo4u is correct. You will need to use the pre-release firmware I developed from this post to work on the Yi Home 27US:  https://github.com/shadow-1/yi-hack-v3/issues/40#issuecomment-315593035

johanneshorak, 2018-01-26:
I have the US47 version of the yihome, can't say how it is for the US27 but I would assume similar. What I noticed is that if your WIFI key is too long, it wont be able to connect. From what I tried I can say that a 24 character password works, while a 28 character password fails, so the limit seems to be somewhere in between.

in summary: if you can't connect the cam to your WIFI you might wanna check if your WIFI password is longer then 24 characters