shadow-1/yi-hack-v3 #92: Support of RTSP anytime soon..?

gojiii, 2017-12-05:
Hello, thanks for making this :) Just wanted to know if a version with RTSP support is incoming anytime soon? Let us know, thanks :)

SeydX, 2017-12-05:
RTSP pleaaaaaaaase

shadow-1, 2017-12-06:
@gojiii
I have been very time poor in the last few months. I already know how I would like to develop RTSP support, however I need to hack some of the Xiaomi programs to determine how they work before I will be able to release something that is stable.

kruschman, 2017-12-06:
Would RTSP support without the Yi app compatibility be any easier? I know that there are a lot of people that would basically like a solution like fritz's that works with the new hardware.

galadril, 2017-12-07:
https://github.com/shadow-1/yi-hack-v3/issues/15

galadril, 2017-12-07:
You can add a bounty on this feature:
https://www.bountysource.com/issues/44753535-rtsp-server

shadow-1, 2017-12-08:
@kruschman
RTSP support without Yi App compatibility might be a little easier to achieve. However it does have its own challenges which are different.

feeling001, 2017-12-14:
Does somebody have an idea how to capture still images "ie : each X seconds via cron job" to put them on the FTP ?
I did'nt find a way to access the device (i suppose trough hi3518e_jpege) to grab images.

regards,
L.

uorriz, 2017-12-14:
I was able to achieve this on my 27US using one of the examples of the SDK. 
The code is in mpp/samples/venc/sample_venc.c
I was able to capture a snapshot and save it as /tmp/sd/snap.jpg (therefore the file snap.jpg is in the root of the SD card) every X ms.
Then I created a symlink to snap.jpg from the www directory. My goal was to implement a sort of crude MJPEG server using the already available HTTP server. Unfortunately, this is not going to work because the HTTP requests could be served while the capture program is saving the jpg file. 
So the next step is to implement an HTTP server directly in the sample program. As soon as I'll have some time I'll work on it.

Obviously this will conflict with the cloud software, so it can only work if the original yi apps are killed.

I'm attaching the binary in case you have a 27US and want to give it a try. To test it, just copy it to the sd card (you can use FTP for that), kill all the yi apps and start it with ./sample_venc 4
[sample_venc.tar.gz](https://github.com/shadow-1/yi-hack-v3/files/1560237/sample_venc.tar.gz)

shadow-1, 2017-12-20:
@uorriz
Great progress! I'm sure this will help many people.

dvv, 2018-01-02:
If we have still images we could use them as the source for https://github.com/jacksonliam/mjpg-streamer/tree/master/mjpg-streamer-experimental/plugins/input_file.

dvv, 2018-01-02:
@uorriz how do we "kill all the yi apps"?

uorriz, 2018-01-02:
@dvv I manually kill them using telnet. 
Here is a script I launch at startup to disable Internet access and kill all the yi apps:

```
#!/bin/sh

# disable internet access
route del default

# kill yi processes
killall watch_process
killall cloud
killall mp4record
killall rmm
killall oss
killall dispatch
killall log_server
killall udevd
killall p2p_tnp
killall arp_test
```

dvv, 2018-01-02:
I get
```sh
/tmp/sd # ./sample_venc 4
=============SAMPLE_COMM_VI_SetMipiAttr enWDRMode: 0
[Func]:HI_MPI_ISP_MemInit [Line]:144 [Info]:ISP[0] get Mem info failed!
SAMPLE_COMM_ISP_Init: HI_MPI_ISP_Init failed!
[SAMPLE_COMM_VI_StartIspAndVi]-2040: SAMPLE_COMM_VI_StartIspAndVi: Sensor init failed!
[SAMPLE_VENC_1080P_JPEG_Thumb]-1623: start vi failed!
program exit abnormally!
```

uorriz, 2018-01-02:
Two things come to mind: 
1) wrong model / sensor. I compiled it to work with the 27US (the 720p model). It will not work with other sensors;
2) yi apps still running;

dvv, 2018-01-02:
The model should match. Any trick to check the model/sensor from telnet session?

dvv, 2018-01-02:
An approach for streaming: `tail -F /tmp/sd/record/tmp.mp4.tmp` would give progressive chunks of mp4 -- if they are of fixed rates then giving them a proper header we could probably make hls (?) based streaming.
Just an idea though

dvv, 2018-01-12:
@uorriz I have
```
/tmp/sd # rtsp/sample_venc
Usage : ./sample_venc <index>
index:
         0) 1*1080p H264 + 1*VGA H264.
         1) 1*1080p MJPEG encode + 1*1080p jpeg.
         2) low delay encode(only vi-vpss online).
         3) roi background framerate.
         4) Thumbnail of 1*1080p jpeg.
         5) svc-t H264
```

but the cam is 720p one.
Couldn't you post here a 720p version of `sample_venc`.

TIA,
--V

ordimans, 2018-01-19:
I am working on RTSP on 47US, i use rtsp2301 .

I have this logwhen i connect to VLC....
`<><><><>Creat RTP<><><><>
pstr=-59927
**adding a schedule object action rtsputils.c,322**
TRANSFER TO READY STATE!
exit Rtsp_state_machine
4
S192.168.0.1 Port:58277
        +++++++++++++++++++++
        start to play 0 now!
        +++++++++++++++++++++
        Start Playing!
exit Rtsp_state_machine
4
SIncoming RTSP connection accepted on socket: 6
EventLoop Connection reached: 4
192.168.0.1 Port:58277
exit Rtsp_state_machine
4
S192.168.0.1 Port:58296
exit Rtsp_state_machine
4
Sfd:4,RTSP connection closed by client.
first error,pRtsp is null
192.168.0.1 Port:58296
server:192.168.0.101
port:-1093458630

GetSdpDescr,703===>psp.base64profileid=,psp.base64sps=,psp.base64pps=

exit Rtsp_state_machine
4
S192.168.0.1 Port:58296
*** transport: RTP/AVP/TCP;unicast;interleaved=0-1 ***
pstr=/TCP;unicast;interleaved=0-1
AAAAAAAAAAA Unsupported Transport,rtspservice.c,1118
TRANSFER TO READY STATE!
exit Rtsp_state_machine
4
S192.168.0.1 Port:58296
*** transport: RTP/AVP;unicast;client_port=59928-59929 ***
!!!!!!!!!!!!!!!!!!!!!!rtp create:addr:100a8c0,port:59928,localc0a80065
<><><><>success creat RTP<><><><>
<><><><>Creat RTP<><><><>
pstr=-59929
**adding a schedule object action rtsputils.c,322**
exit Rtsp_state_machine
4
S192.168.0.1 Port:58296
        +++++++++++++++++++++
        start to play 0 now!
        +++++++++++++++++++++
        Start Playing!
exit Rtsp_state_machine
4
S^Cbegin free mem
The Server quit!
/tmp/sd/test/app # ^C
`

kapollo, 2018-03-23:
Hi I don't know what exactly rtsp2301 is under the hood, but I do own a D-Link 5222L B2 which is based on Hi3518C, and it has RTSP streaming based on the Live555 media server. 
Dlink like some other OEMs has gpl code for their gpl sw, and for cameras at least it equals to a buildable fw somewhat (it doesn't contain rootfs, but then neither does their fw updates). 
All in all there is code for Live555 which already has some interfacing done with the SDK binaries from the chipmaker. It could be a great starting point. At least it should be fairly easy to combine venc with stills and mjpeg stream for starters.

There is also other apporach possible like in Edimax's cameras it uses spook as RTSP server (GPL code also provides source code for spook which interfaces with the raw stream directly).
hope this helps,  just had some encounters with both and wanted to add my two cents.

pmarrapese, 2018-03-24:
Hi all,

Over the last few weeks, I've been reverse engineering the Yi desktop app. This has enabled me to begin figuring out the underlying protocols used to communicate with the camera.

I have reached a point where I am able to discover a camera on the LAN, perform the handshake, start receiving audio/video, and successfully process (decrypt and decode) the streams. Additionally, I can issue commands such as PTZ control and quality control to the camera. I have started working on protocol documentation and a Node.js library which I will use as the backbone for RTSP and HTTP interfaces to the camera.

I am confident this will greatly open up development potential for Yi cameras. Stay tuned...

dwild, 2018-04-04:
@pmarrapese I guess that means you will still depends on the cloud software running? Will it still works without internet connection? I'm really eager to see what you worked on!

pmarrapese, 2018-04-04:
This will not depend on the cloud software -- it is a completely independent protocol re-implementation. :)

There are cases where an internet connection is required even for a peer-to-peer LAN connections. The reason for this is that the camera generates a new password upon each boot, which is directly transmitted to the Yi cloud API and only obtainable from there. That being said, if that password is obtained once & the camera hasn't cycled it, a LAN connection would be possible without an internet connection.

dwild, 2018-04-04:
@pmarrapese When I said cloud software, I was thinking on the camera.

Sadly where my camera will be setup, there may not have internet connectivity all the time and a power cycle may happen while there's no internet connection. How many API methods does theses call requires? Do you think it would be possible to re implements those? If it's only to store the camera passwords, that may not be that many I guess and I could re implements those easily. I could then use a local DNS server to redirect the call to a local server or even probably edit it on the camera itself (the domain is probably simply a string in the cloud software).

Do you think you may release your finding soon? I look foward to work on that setup!

Thanks again for all theses efforts!

pmarrapese, 2018-04-04:
@dwild Ah, I see! Yes, this will work with either stock firmware or yi-hack firmware.

The password is accessible on the camera filesystem. So, for your case, the simplest way to retrieve it would likely be through SSH/telnet. For example, a script could scp the required file from the camera, parse the password out, and start the RTSP server with it. I can look into supporting this.

dwild, 2018-04-04:
@pmarrapese That's even better! You are amazing, thank you!

You don't have to support it for now. As long as I can give the password of the camera to your software, it should be fine to me (and considering you are planning to do it in Node, that going to be easy for me).

dvv, 2018-04-05:
@pmarrapese can we look at protocol current documentation?

JocPelletier, 2018-04-05:
What happened to @pmarrapese messages about his progress?

dwild, 2018-04-05:
He is afraid of infringing the ToS of Xiaomi.

DarthMadeHer, 2018-07-30:
I have copied the yi-vencrtsp file from the SD directory onto a working SD that has shadow-1's firmware hack. My understanding is that the startup.sh will call what it needs to from the yi-hack-v3 dir on the SD card. Do I need to telnet to the camera and manually start the processes? Is the url the same as previous ones rtsp://root:@IP_Address:554/ch0_0.h264?

eldinh, 2018-08-12:
@DarthMadeHer so what happens after you cp manually, is rtsp working now? What is the url? Thankx

jpenyc, 2018-08-13:
I used this https://github.com/Necromix/yi-vencrtsp on this YHS-113-IR (27US) and it works using vencrtsp
How do I get vencrtsp to start without having to ssh in.

Minims, 2018-08-13:
@jpenyc Can you explain how you make it working ?  I try to follow the readme on my 27US cam, but i may do something wrong. Thank you !

bobaxx, 2018-08-29:
Hello. For those who are in the same situation as me: waiting for RTSP on the 27US camera for a looooooooong time.
You can do it mixing the two following firmwares:
*yi-hack-v3 from Shadow1
*yi-vencrstp

Here are the steps (it worked for me on 2 different cameras). It's a bit tidious but it works:

1-Copy the files from the "SD" directory from necromix firmware (https://github.com/Necromix/yi-vencrtsp) to the root of your SD card;

2-Flash your camera using yi-hack-v3 and following the steps on the page : https://github.com/shadow-1/yi-hack-v3;

3-After the camera is flashed successfully, you should acess it with its IP in your web browser. For example , mine is 192.168.1.65. 

4-Go to "system config" tab and enable telnet server, then apply. Your camera will reboot.

5-Connect to your camera with telnet using its IP, with putty for example.
 Port is 23, user login is "root", and password is empty.
 For now, you should be logged on your camera.

6-Use the following commands:
   cd /tmp/sd/yi-hack-v3/
   ./startup.sh

  This script will replace some native properties of the original camera firmware.
  If the script doesn't work (it can happen), you have to do it manually using this command line:

  cp /tmp/sd/yi-hack-v3/yi_init.sh /home/app/init.sh

7-After camera reboot, you should be able now to use RTSP ! 
   Unfortunately, you have to start the service manually but hey, it's a good start !

  To launch RTSP service, connect to your camera with telnet (as seen before) and type this lines:

cp /tmp/sd/
./vencrtsp

And voilà ! You can now access RTSP with VLC or another client using rtsp://root@CAMERAIP:554/ch0_1.h264. Mine is rtsp://root@192.168.1.65:554/ch0_1.h264

I noticed a bug when the ambient light is too low. The rtsp service show error "segmentation fault" and stops working.

Enjoy !

McVillano, 2018-08-30:
@bobaxx would the original app still work after this mod?¿

Minims, 2018-08-31:
RTSP seems working, but  i have sometimes some segfaults on my 27US. 
The original app doesn't work anymore on my side. but you can revert by reinstalling the original yi-hack-v3

McVillano, 2018-09-01:
Yep, finally i´ve turned back my camera, i was lookingfor full functional RTSP and the project has been without update for months i need rtsp for Home Assistant

fluidof, 2018-09-14:
@bobaxx I'm arrived at the step below, why I cannot access via VLC? thanks
[image]
[image]

fluidof, 2018-09-14:
@bobaxx but just for my understanding, after starting the service with ./vencrtsp shall I have to leave open the SSH connection? Can I close terminal? It seems not...
[image]

fluidof, 2018-09-15:
@bobaxx And, is there a way to run a command at startup to run automatically the ssh root@xxx.xx.xxx. ./tmp/sd/vencrtsp or something similar?

hoopsta1423, 2018-10-25:
@bobaxx This guide worked perfectly for me so thank you! Has anybody had any luck using this with BlueIris? I can connect with VLC but can't get BlueIris to pull the stream in using the rtsp address

jpawlowski, 2018-11-27:
That wouldn't work anymore it seems because the kernel modules are not compatible with newer firmware versions. I opened a bug report for it: https://github.com/Necromix/yi-vencrtsp/issues/7

Necromix, 2018-12-02:
@fluidof: to start vencrtsp at startup.

Edit the /tmp/sd/yi-hack-v3/yi_init.sh or /home/app/init.sh

Add to line 58 (just before the exit)
`/tmp/sd/vencrtsp &`

Necromix, 2018-12-02:
@jpawlowski 
I anticipated this issue and in my initial plan, I would like to use the original ko from CAM but I never achieve to run yi-vencrtsp on those modules.

We can try [https://github.com/Necromix/yi-vencrtsp_v2](url) which uses all files from cam and extract video from /tmp/view.

mvklingeren, 2019-01-01:
That is the correct url.

bobaxx, 2019-05-15:
It's a bite late, but the correct adress is
rtsp://root@ipaddress:554/ch0_0.h264

plutosherry, 2020-11-04:
`/home/yi-hack-v3 # ./vencrtsp 
RTSP:-----Init Rtsp server
=============SAMPLE_COMM_VI_SetMipiAttr enWDRMode: 0
[Func]:HI_MPI_ISP_MemInit [Line]:144 [Info]:ISP[0] get Mem info failed!
SAMPLE_COMM_ISP_Init: HI_MPI_ISP_Init failed!
[SAMPLE_COMM_VI_StartIspAndVi]-2040: SAMPLE_COMM_VI_StartIspAndVi: Sensor init failed!
[SAMPLE_VENC_1080P_CLASSIC]-1351: start vi failed!
program exit abnormally!`