shadow-1/yi-hack-v3 #93: all yi dome camera offline??

fredyagu, 2017-12-08:
Hi, any body got problem like mine?Since yesterday, all my dome camera status turn to offline, and it says offline at 2017/12/07 07:04. Any solution?

shadow-1, 2017-12-08:
@fredyagu
Try using the "Test Config" feature on the web interface. Maybe the configured proxy servers are no longer working and you need to find new proxy servers. 

This is assuming you are bypassing the Mainland China region lock.

fredyagu, 2017-12-08:
I've tried to reset and pairing it again , it is able to view but after a couple minutes, it will turn offline again.

fredyagu, 2017-12-08:
@shadow-1
Please try your dome camera, is it still working or not?

shadow-1, 2017-12-08:
@fredyagu
I no longer have a Chinese version of the camera so I can not check this functionality.

I know others have had issues in the past. I might need to check the closed threads.

From memory, there have been issues if you use an old version of the Yi Home app. Make sure you have the latest version of the app installed on your phone.

genz10, 2017-12-09:
Yes, I'm also experiencing this issue. Detail how to reproduce :

1. Successfully Connect 13CN to latest Yi Home App
2. Close Yi Home App for couple minutes
3. Re-open Yi Home App, and camera will be displayed "Offline At : XXXXX"

If the application is not closed, camera will able to be operated normally. 

I've tried to enable VPN (Singapore based) when open Yi Home App and it still not working.

shadow-1, 2017-12-09:
@genz10 
Isn't the 13CN version of the camera the original Yi Home?

Hopefully more people can contribute so we can work out what could be the potential cause of the issue.

genz10, 2017-12-09:
@shadow-1 

Sorry, I don't get your question. 

Is it possible xiaomi blocked all region outside china for camera that only can be used within China.

art3ar, 2017-12-10:
Same issue here

shadow-1, 2017-12-10:
@genz10
Sorry for confusing. I was just wondering what camera you were using.

@fredyagu @genz10 @art3ar
Can you post a screenshot from the "Test Config" page when the test has completed.

I have confirmed with a Yi 1080p Dome camera (Chinese version) that everything is working as expected.

art3ar, 2017-12-11:
[image]

fredyagu, 2017-12-11:
[image]

fredyagu, 2017-12-11:
Can we use proxychain in p2p binary? If yes, can you tell me how to do it? From my analyze the p2p_tnp use udp port 32100

shadow-1, 2017-12-11:
@art3ar 
I have tested your proxy server and it appears to be working properly and appears to be based in Mainland China. Unfortunately I can't do a proper test as I don't have a camera that is locked to Mainland China.

It is a long shot, however you can try disabling some services temporarily and see how it goes. You can also try using a different power adaptor. For example try using the one that came with your phone rather than the one that comes with the camera. There have been at least two people where this has managed to resolve their issues. Although the camera stopped working after a few hours rather than minutes and everything stopped working including the web interface.

@fredyagu 
I don't think using ProxyChains-ng will work with the p2p traffic. ProxyChains-ng only supports TCP communications. It will not work with UDP. You may be able to do something like what you want if you use a computer as a router for the camera communications along with a SOCKS proxy. Perhaps with a bit of tinkering a router that supports DD-WRT or a MikroTik router could be used.

genz10, 2017-12-11:
@shadow-1 

I'm currently doesn't have access to my camera, but I've run test config yesterday and all looks good. My camera also displays as @fredyagu posted. 

For additional info, here's what i've already tried :

- Using singapore VPN connected to phone which Yi Dome connected to. 
- Change socks5 IP address.

fredyagu, 2017-12-12:
I think dome cameras blocked through the p2p connection. Anyway to bypass it?

lmlbenny, 2017-12-13:
i have same problem to. i'm already restart, upgrade, downgrade and change proxy. no one can solve the problem.

when i reset the camera and pair again. it's can use. but when you out from live streaming and refreshing camera list. my camera will be back to offline. datetime after before and after reset is same.

anyone can help us?

[image]

shadow-1, 2017-12-13:
I find this issue quite perplexing! I asked a friend who uses the Chinese version of the camera whether he has any issues and I was told that everything works perfectly.

@lmlbenny @fredyagu @genz10 @art3ar 
What camera are you using? Maybe you can post the log files which automatically get saved to the microSD card.

Also, do any of you have serial port access? If not, I might have to come up with a way to record the console logs through Telnet.

lmlbenny, 2017-12-13:
the camera type is YI DOME 720P Serial : 13CNYPX201.

i have try downgrade the application YI Home. that still not working.
- i clean all junk file, cache and uninstall Yi Home.
- reinstall Yi Home.
the camera can work when all junk/cache clean. when back and refresh camera list, the camera became offline again just like when I reset the camera with same datetime.

[log_start_reset.tar.gz](https://github.com/shadow-1/yi-hack-v3/files/1555181/log_start_reset.tar.gz)
[log_wifi_connected.tar.gz](https://github.com/shadow-1/yi-hack-v3/files/1555182/log_wifi_connected.tar.gz)
[log_bind_fail.tar.gz](https://github.com/shadow-1/yi-hack-v3/files/1555183/log_bind_fail.tar.gz)
[log_bind_success.tar.gz](https://github.com/shadow-1/yi-hack-v3/files/1555184/log_bind_success.tar.gz)
[log_first_login.tar.gz](https://github.com/shadow-1/yi-hack-v3/files/1555185/log_first_login.tar.gz)

fredyagu, 2017-12-13:
It begins in 7 december where suddenly all the 13CN dome camera came to offline. Then in 11 december the 43CN dome also having the same problem.
I can pair the camera with the app, and it is only connected locally, and after a couple minutes the camera in the app will turn offline.
I can say that it is connected locally, because if i already pair it then i change my phone to use 4g, it will say error 3006, then when i connect again with the same wifi, it can view again.

shadow-1, 2017-12-13:
@fredyagu 
Hmm, very interesting. I'm still confused because all the newer Yi cameras operate in exactly the same way. So if the Yi Dome stopped working because of something Xiaomi implemented on their server. I expect that all the Yi cameras which I currently support to stop working as well.

@lmlbenny 
When you delete all the tar.gz files from your microSD card and restart the camera. Does `log_bind_fail.tar.gz` get recreated?

lmlbenny, 2017-12-13:
log_bind_fail.tar.gz created when i had downgrade to version 0.1.2.

lmlbenny, 2017-12-13:
if i'm use version 0.1.3 log_bind_fail.tar.gz never created

fredyagu, 2017-12-13:
I think dome cameras blocked through the p2p connection. maybe checking serial number
you can try kill p2p_tnp ants 720p 17CN, and camera will offline -3006

lmlbenny, 2017-12-15:
this i using serial com to see progressing

when i restart 
[image]

lmlbenny, 2017-12-15:
after scan barcode :
[image]

after pair success 
[image]

lmlbenny, 2017-12-15:
when live streaming camera 
[image]

lmlbenny, 2017-12-15:
when i rotate the camera

[image]

lmlbenny, 2017-12-15:
after go out from live streaming.
and refresh the camera list 
[image]

lmlbenny, 2017-12-15:
System startup
hello, open console

 do_auto_sd_update: detect flash size: 0x1000000, gsdexist=0

 do_auto_update: detect flash size: 0x1000000
Hit any key to stop autoboot:  0
16384 KiB hi_fmc at 0:0 is now current device

## Booting kernel from Legacy Image at 81000000 ...
   Image Name:   7518-hi3518c-kernel
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1384824 Bytes = 1.3 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0
Linux version 3.4.35 (chenshibo@ANTS-SH-SV02) (gcc version 4.8.3 20131202 (prere                                                                                        lease) (Hisilicon_v300) ) #1 Wed May 25 19:18:17 CST 2016
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: hi3518ev200
Memory policy: ECC disabled, Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 5588
Kernel command line: mem=22M console=ttyAMA0,115200 root=/dev/mtdblock4 rootfsty                                                                                        pe=jffs2 mtdparts=hi_sfc:256k(boot)ro,64k(env),64k(conf),1600k(os),1280k(rootfs)                                                                                        ,12992k(home),64k(vd1),64k(ver)
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 22MB = 22MB total
Memory: 18480k/18480k available, 4048k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0xc1800000 - 0xff000000   ( 984 MB)
    lowmem  : 0xc0000000 - 0xc1600000   (  22 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0340000   (3296 kB)
      .init : 0xc0340000 - 0xc035cc64   ( 116 kB)
      .data : 0xc035e000 - 0xc037f220   ( 133 kB)
       .bss : 0xc037f244 - 0xc03b1850   ( 202 kB)
SLUB: Genslabs=13, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:32
VIC @fe0d0000: id 0x00641190, vendor 0x41
sched_clock: 32 bits at 49MHz, resolution 20ns, wraps every 86767ms
Console: colour dummy device 80x30
Calibrating delay loop... 298.59 BogoMIPS (lpj=1492992)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
Initializing cgroup subsys freezer
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x8026ab08 - 0x8026ab60
dummy:
NET: Registered protocol family 16
Serial: AMBA PL011 UART driver
uart:0: ttyAMA0 at MMIO 0x20080000 (irq = 5) is a PL011 rev2
console [ttyAMA0] enabled
uart:1: ttyAMA1 at MMIO 0x20090000 (irq = 30) is a PL011 rev2
uart:2: ttyAMA2 at MMIO 0x200a0000 (irq = 25) is a PL011 rev2
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Switching to clocksource timer0
NET: Registered protocol family 2
IP route cache hash table entries: 1024 (order: 0, 4096 bytes)
[... 861 lines cut ...]
g_dispatch_info.mmap_info->ptz_x_runtime =4 2
p2p cur user 0 SessionHandle(2)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(8)
p2p cur user 0 SessionHandle(2)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(8)
p2p cur user 0 SessionHandle(2)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(8)
p2p cur user 0 SessionHandle(2)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(8)

~ # p2p cur user 0 SessionHandle(2)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(8)
p2p cur user 0 SessionHandle(2)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(8)
g_dispatch_info.mmap_info->ptz_x_runtime =4 2
p2p cur user 0 SessionHandle(2)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(8)
p2p cur user 0 SessionHandle(2)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(8)
p2p cur user 0 SessionHandle(2)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(8)
p2p cur user 0 SessionHandle(2)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(8)
p2p cur user 0 SessionHandle(2)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(8)

lmlbenny, 2017-12-15:
second time reset and pair again with SD Card 

*** Board tools : ver0.0.1_20121120 ***
[debug]: {source/utils/cmdshell.c:166}cmdstr:himm
0x2003002c: 0x000C4003 --> 0x001C4001
[END]
acodec inited!
insert audio
mipi_init
init phy power successful!
load hi_mipi driver successful!
==== Your input Sensor type is ov9712 ====
*** Board tools : ver0.0.1_20121120 ***
[debug]: {source/utils/cmdshell.c:166}cmdstr:himm
0x201200cc: 0xFF035144 --> 0xFE033144
[END]
*** Board tools : ver0.0.1_20121120 ***
[debug]: {source/utils/cmdshell.c:166}cmdstr:himm
0x201200c8: 0x00F22C2C --> 0x00023C2E
[END]
*** Board tools : ver0.0.1_20121120 ***
[debug]: {source/utils/cmdshell.c:166}cmdstr:himm
0x201200d8: 0x1E1EC001 --> 0x0D1EC001
[END]
Load hi_cipher.ko success.
[CPLD_PERIPH] Timer closed.
[CPLD_PERIPH] Timer closed.
[CPLD_PERIPH] Timer opened.
killall: recbackup: no process killed
killall: lua: no process killed
sensor_register_callback ok
linear mode
=========================================================
===ominivision ov9732 sensor 720P30fps(Parallel port) init success!=====
=========================================================
********************* Open pwm device error! *********************
 /tmp/view: memory clear 781120
MMAP address=0xb658e000, size=781120(0xbeb40), mode=0, try 1
Hisilicon Watchdog Timer: 0.01 initialized. default_margin=60 sec (nowayout= 0, nodeamon= 0)

[PID_LIST] pid_list_init ok, [ ver=Jan 16 2017, 14:30:19 ]
WATCHDOG @Hi3518v200
WATCHDOG @Hi3518v200
tcpsvd: listening on 0.0.0.0:21, starting
[CPLD_PERIPH] Timer closed.
[CPLD_PERIPH] Timer opened.
Failed to connect to non-global ctrl_ifname: wlan0  error: Conne[CPLD_PERIPH] Timer closed.
ction refused
[CPLD_PERIPH] Timer opened.
ash: 10: unknown operand
0x1300 = 00064300
Number of Packet Allocated in open = 1
Number of Packet Freed in open = 0
killall: wpa_supplicant: no process killed
killall: udhcpc: no process killed
ash: 1: unknown operand
supp-global-ctrl-iface-init (will try fixup): bind(PF_UNIX): Address already in use
killall: udhcpc: no process killed
rm: can't remove '/tmp/resolv.conf': No such file or directory
[CPLD_PERIPH] Timer closed.
cmd(rm /tmp/sd/log_wifi_connected.tar.gz;rm /tmp/sd/wifi_connected -r;mkdir /tmp/sd/wifi_connected;ps > /tmp/sd/wifi_connected/ps1.txt;cp /home/homever /tmp/sd/wifi_connected;lsmod > /tmp/sd/wifi_connected/lsmod.txt;ls -laSR /home > /tmp/sd/wifi_connected/file.txt;cp /tmp/wpa_supplicant.conf  /tmp/sd/wifi_connected;cp /tmp/log.txt /tmp/sd/wifi_connected;df /home > /tmp/sd/wifi_connected/df.txt;df / >> /tmp/sd/wifi_connected/df.txt;tar -cf /tmp/sd/log_wifi_connected.tar.gz /tmp/sd/wifi_connected;rm /tmp/sd/wifi_connected -r &)
rm: can't remove '/tmp/sd/wifi_connected': No such file or directory
tar: removing leading '/' from member names
killall: mp4_record: no process killed

rmm.c(ai_get_lpcmstream-6017)[05:47:18.718]:pid[1173]

rmm.c(ai_get_lpcmstream-6018)[05:47:18.718]:AiGetLPCMStream start

rmm.c(ai_get_aacstream-6168)[05:47:18.747]:pid[1174]
[... 90 lines cut ...]
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading /lib/libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Random chain  ...  110.80.1.54:1080  ...  54.244.20.70:80 trans_json_ex result=20000
trans_json_ex result="ok":"false"
trans_json_ex result=false
 ...  OK
p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(2)
[CPLD_PERIPH] Timer closed.
[CPLD_PERIPH] Timer opened.

MMAP address=0xb67d3000, size=781120(0xbeb40), mode=0, try 1
MediaDataGetNextFrame_withlen: ch 1, bdindex  20, framesize 0x  43e4 > 0x 60000/0x  27b4, or < 0x5 error
killall: recbackup: no process killed
p2p cur user 0 SessionHandle(1)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(2)
killall: lua: no process killed
x =1403 656 1475 600 x_step =-72 56 x_step_2 =0 0
MediaDataGetNextFrame_withlen: ch 1, bdindex  79, framesize 0x  5057 > 0x 60000/0x  47e4, or < 0x5 error
p2p cur user 0 SessionHandle(1)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(2)
MediaDataGetNextFrame_withlen: ch 1, bdindex 121, framesize 0x  5695 > 0x 60000/0x  5454, or < 0x5 error
p2p cur user 0 SessionHandle(1)
p2p cur user(1) max_user_num(1) in_packet_loss(0) out_packet_loss(2)
p

shadow-1, 2017-12-17:
@lmlbenny 
Thanks for the logs. Nothing seems to jump out at me. Although something went wrong with the first log and the camera restarted itself through the watchdog application.

Can you try downgrading the official Xiaomi firmware. Make sure you are using the latest version of yi-hack-v3 to ensure the custom software I wrote will not be wiped by the Xiaomi updates.

Once downgraded, can you check whether the same issue occurs. If it does not, you can try upgrading back to the latest firmware and see if the problem resurfaces.

I have hosted all previous Xiaomi firmware files (that I could get hold of) on the following page: https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

fredyagu, 2017-12-18:
dome 13cny network traffic summary
[image]

lmlbenny, 2017-12-18:
@shadow-1 
still not work. i had downgrade and upgrade to 0.1.4 Beta 2.

i see this command in serial com
trans_json_ex result=20000
trans_json_ex result="ok":"false"
trans_json_ex result=false

lmlbenny, 2017-12-18:
[image]

FumaKilla, 2017-12-19:
my 13CN yidome has exactly the same problem... 
offline at 2017/12/07 07:04
and got error 3006 after restarting yihome..

been trying to refresh the proxy list changing them to at least 6 proxies that got checks mark. 
changing firmware to stock.. resulted to camera locked it self..... 
changing to your latest 4 release..... still no luck.... 
is there anything. I can  help you to resolve this problem?

FumaKilla, 2017-12-19:
I'm using the provided yidome power adaptor
has been changing yihome version too has no luck... 
how ever I'm still be able to receive notifications from the camera (motion detector)...

shadow-1, 2017-12-20:
@FumaKilla
Can you confirm what version of Yi Home you are using? Almost exactly the same problem has occurred before due to using an old version of Yi Home.

At that time I ended up uploading an APK file to the latest Android version.

The issue appears to be different this time around. And I am puzzled why it only affects Yi Dome cameras and none of the other supported cameras. They are all essentially the same.

lmlbenny, 2017-12-20:
i have use my new yi dome. i never use before and this is the result.

lmlbenny, 2017-12-20:
[image]

i will be crazy right now.

fredyagu, 2017-12-20:
@lmlbenny  13CNY too?

lmlbenny, 2017-12-20:
@fredyagu yes.

FumaKilla, 2017-12-21:
@shadow-1 
i'm using Yihome version 2.10.15.0_20170918 
this is the one that give "camera is offline, please reconnect it (-3006)"
other version just show my camera is offline at 2017/12/07 07:04
even the newest one from Yi

is it possible that they put something on the server to block YiDome13CN  version only? because AFAIK this YiDome is the most wide spread hacked IP camera that originally made strictly for the main land so it took the most bandwidth from their server?

shadow-1, 2017-12-21:
@FumaKilla 
The latest version of the Yi Home app is: 2.10.19.1_20171213
Can you try this version and see if the issue gets resolved?

It is possible that Xiaomi could have found a way to get around my hack on their server and only implement it on certain cameras. However I find it strange as the Yi Dome 1080p, Yi Home 1080p & Yi Home 17CN and all essentially the same as the Yi Dome. It would be trivial for Xiaomi to block everyone!

What you are describing about the extra security layer is exactly what Xiaomi implemented and I managed to get around it using proxy server. It is difficult to defeat this on the Xiaomi server side. However it is quite easy to defeat via firmware update.

spunz84, 2017-12-26:
Hi, i am just a person who has the same problem with my yidome mentioned in this thread. I was googling the solution and bumped into this. It seems like I am nt the only one. But something interesting has happened that might help. I hv three yi dome cameras all 13CN. The person I bought from somehow helped me and I cud view the cameras only if my phone is connected to the same wifi network the cameras are connected to. However, since yest i am able to view one of my yi dome via data not need wifi I can access it wherever I am! The other one I can view via wifi but not wid my data plan. And the other one have a problem pairing. The one which I can view via my phone data plan gives me (and I hope you all) a glimmer of hope. I hope this helps, i wish I could contribute more but like I said I am no programmer or hacker. Just a user. If you wud like me to check for firmware or anything I wud be happy to do so. However please be kind enough to guide me on how to do it, so I can give you feedback. Thank you.

fredyagu, 2017-12-26:
@spunz84
Can you tell me, the serial number under the camera qrcode is begin with 13CN or else?
Because i found that the one that has serial else then 13CN are working fine now. Although when we scan the qrcode, it return 13CN serial. 
Btw, where do you live?

FumaKilla, 2017-12-28:
@shadow-1 nope.... the version you mentioned only give the camera offline at 2017/12/07 07:04 status.. 
but it still send notification on motion tracking... and I'm able to view the recorded video.  so somewhat the camera is still useful for me... 
only the online viewing is missing....

FumaKilla, 2017-12-28:
good job on the hack though :) thank you for the firmware...... with out it the camera will only said can only be used in china and locked it self..... I salute you for that.

lmlbenny, 2017-12-28:
i got solution for a while.

the solution is :
- unistall the yi home app and install yi home handler mod.

this application can help me connect my yi dome with local connection (same wifi with the camera) but can't access from outside.

lmlbenny, 2017-12-28:
https://www.dropbox.com/s/l2sdwb6s0tkfywi/YiHomeCN2.10.15.0_20170918*apk?dl=0

shadow-1, 2017-12-30:
Great discussions here! It appears that this is a real issue.

Unfortunately I think I can only provide limited advice as I do not have an affected camera to closely examine the cause and find potential solutions.

However the logs that @lmlbenny provided indicate some kind of proxy server issues.

@FumaKilla 
Do you happens to have serial port access to post logs which are displayed on the console? If you don't, we may be able to manage with just Telnet/SSH. I just want to determine if you are getting the same error messages like @lmlbenny .

wikeaj, 2018-01-06:
any solution to this problem? mine also is seen as offline

FumaKilla, 2018-01-14:
@shadow-1 I do not have serial port acess to it... nor have idea how to telnet/ssh... I'm noob at it... xD willing to though..  is there any tutorial I can follow?

agenonlinesamarinda, 2018-01-16:
i have same problem to. i'm already restart, upgrade, downgrade and change proxy. no one can solve the problem.i have same problem to. i have a yi dome 1080p 15CN, 
so far i use yihome PC china download at http://publicfiles.xiaoyi.com/yi_files/YIHomePCClient_Download_1.0.0.2_2017-04-07.exe
and using yihome android just like @lmlbenny used.
[image]
[image]
[image]
all can only work on wifi locale

genz10, 2018-01-18:
I found that some tricks can make Yi Dome work again,

Try to back date some days before date shown at "Offline At: XXXXXX"

fredyagu, 2018-01-19:
@genz10
Can you explain in detail how to do it?

aosmd, 2018-01-28:
up

spunz84, 2018-02-14:
So thats it then? My yidome china version has become trash. The hacker that made it accessible for outside china cannot be bothered to fix this problem. If only v knew that it was made accessible by a lazy hacker who cud nt be bothered to find a solution because he apparently doesnt have the problematic equipment in his hands at the moment. So much for accountability. Lesson learned. Never buy things that can only operate when hacked. Especially if it is hacked by someone who is too lazy and preoccupied to fix d damn thing. Getting an imternational version. Unfortunately my supplier also does not seem to be accountable as well. So this yidome is probably gonna be stuck in some poor dolphin's neck one day.

andyg2, 2018-02-15:
@spunz84 
ok bye then, have a nice life somewhere else.

totobear, 2018-03-01:
here's my 2 cents worth:  I just added 2 yi home 720p dome cameras to 2 others and they all went offline periodically despite resets.  sometimes 1 or 2 would be online but then also go offline.  it turns out my brother had upgraded the network from optimum to verizon fios and reconfigured guest wireless with the same ssid and password expecting the cameras to connect, which they did sporadically.  however, fios guest wireless is 2.4ghz/5.0ghz and it was causing the cameras to go offline.  i disabled guest wireless and configured a 2.4ghz network on a separate channel and all is good now.

FumaKilla, 2018-03-07:
Thanks but  It's not network problem...
@shadow-1 is there any hope for the 13CN?

[10 more replies]