shadow-1/yi-hack-v3 #94: Yi Home IOS App

Deee46, 2017-12-08:
Thank you shadow for the hack. now my camera is finally usable after a month.

quick question:

1: for the love of me i cannot download IOS app, when i switch to china store it wont let me download the app. is there a direct download or something to bypass itunes?
2: is it safe to upgrade firmware? right now i see a new one with a version C in it
3: any chance i can me "Mi Home" app works with this camera? i also have the xiaomi Mijia app and it would be nice to just use one app instead of 2.

Thanks in advance and once again Thank you for sharing.

ykhandler, 2017-12-09:
1. Find (yismart) in appstore after changing your region to china, select the one with chinese letter to download.
2. Yes if you use the newest hacks.
3. I think it will not, it is xiaoYi's not xiaoMi's.

shadow-1, 2017-12-09:
@Deee46 
@ykhandler is correct.

1. I do not have an iPhone. Android is a lot easier as you can manually force the installation of an APK file manually. The advise @ykhandler has provided sounds reasonable.

2. As long as you have installed yi-hack-v3 0.1.4-beta2 or newer, updating the Xiaomi firmware is safe.

3. This is up to the developers working for Xiaomi. I do not develop the smartphone app at all. The Mi Home app was developed specifically for the MiJia range of devices whilst Yi Home was developed specifically for the Yi range of devices. Having said this, as far as I am aware, Xiaomi developed a plugin which provides Yi Home support with the Mi Home app.  Hopefully Xiaomi develop further plugins for the other Yi devices.

fredyagu, 2017-12-12:
Login with China Account then open this
https://itunes.apple.com/cn/app/xiao-yi-zhi-neng-she-xiang-ji/id931168730

Deee46, 2017-12-13:
thank you.. made it to work finally!!

last question, any chance to enable NAS recording on this camera or modify "MI WIFI Hard disk" to work in my netgear N7000.

Thanks.

shadow-1, 2017-12-13:
@Deee46 
That is great news that everything is working!

Unfortunately proper NAS recording is not supported. However I know some people have managed to write a script which automatically downloads the recordings that the camera saves to the microSD card to their NAS or to cloud storage (Gooogle Drive, Dropbox etc). Have a look in this thread: https://github.com/shadow-1/yi-hack-v3/issues/13

Deee46, 2017-12-13:
@shadow-1 - exactly what i needed.. thank you for the link and for the help making this camera work for me.

Deee46, 2017-12-13:
anyone can help me figuring this out? https://github.com/AndMu/YiScanner/issues/3

i tried the Yiscanner and edited the settings to reflect my yi ftp (192.168.1.5) and just not doing anything. perhaps im missing something?

TIA.

mikeschi, 2017-12-14:
Can someone tell me the current version of YI Home China app? (Profile --> About). Mine is 2.13.1

Deee46, 2017-12-14:
Mikeschi

Mine is 2.19.0 but not sure if this is the latest one.

shadow-1, 2017-12-17:
@mikeschi 
I just checked the latest version of Yi Home (Chinese version) for both iOS and Android.
iOS: 2.19.0
Android: 2.10.17.1

shadow-1, 2017-12-17:
@Deee46 
I have not used YiScanner before. Firstly, check that the video files are being recorded to your microSD card. You may need to enable the "activity detection recording" feature within the Yi Home app.

ykhandler, 2017-12-17:
Android latest version is: 2.10.19.1_20171213