shadow-1/yi-hack-v3 #96: Replace telnet/ftpd with dropbear

dvv, 2017-12-17:
Hello!

May I suggest to replace `telnetd` with a tiny SSH server?
[Here](http://landley.net/aboriginal/downloads/old/binaries/1.4.3/extras/dropbearmulti-armv5l) we have the prebuilt binary which is both server `dropbear` and client `scp`.
Not only SSH may replace both insecure `telnetd` and `ftpd`, but one can securely execute remote commands, forward ports, transfer files etc.

TIA,
--Vladimir

shadow-1, 2017-12-17:
@dvv 
Yes we can cross-compile `dropbear` and use that instead of `telnetd` and `ftpd`.
The file size needs to be kept to a minimum as the camera has very limited free space.
This will most likely mean removing some features at compile time to reduce to final executable file size.

However the linked pre-built binary is quite small as long as we don't need to provide any additional libraries which are not included on the camera.

dvv, 2017-12-17:
Right. I would also like to have a fatter busybox -- for example it has barebone `sendmail`. May be you might build one with `allyesconfig` statically to see the binary size. Those who run the camera with SD card would have no problem to store additional stuff there.
TIA

shadow-1, 2017-12-17:
@dvv 
The rootfs partition (partition with Busybox executable) is 1MB in size total. At the moment there is approximately 152KB of free space with the very limited Busybox build.

There is a little more free space on the home partition.

However you are absolutely correct that the microSD card can be used for additional storage for programs.

dvv, 2017-12-17:
My rationale is simple: `/bin/busybox` is about 480k, rootfs (if considered read-only as it imo should be) keeps 150k of otherwise completely _unused_ space, which should allow for much more complete set of system tools which in turn would allow for reacher hacking ;)

shadow-1, 2017-12-17:
@dvv 
I agree. We just need to be a bit selective with the utilities.

I will try cross-compiling dropbear and see how large it is. I don't see any issues with including sendmail.

dvv, 2017-12-22:
I wonder if you could point to where from to get rootfs/homefs for the changes you made?

shadow-1, 2017-12-22:
@dvv 
To get a starting point for the rootfs and home partitions, you will have to extract a firmware image.

If you would like to start with a clean unaltered image, you can extract one of the Recovery images available from: https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr
If you would like to start from one of my firmware images, they are available here: https://github.com/shadow-1/yi-hack-v3/releases

I have written instructions on how to create firmware images in this post: https://github.com/shadow-1/yi-hack-v3/issues/23
I have had to change the layout of the software a little bit due to restrictions on the Yi Dome (I assume Yi Home 17CN/27US/47US have the same issue) along with further improvements (mainly the ability for the software to recover from a Xiaomi software update). However the guide is still 90% accurate.

ykhandler, 2017-12-23:
@dvv, you dont need to put the binary in rootfs partition, you may put it in home partition, then to symlink it to the /bin or /sbin.

dvv, 2017-12-23:
I thought you'd made a release for this.

shadow-1, 2017-12-24:
@dvv 
I have just created a new release with Dropbear and a few other improvements.

dvv, 2017-12-24:
Great, thank you!
I quickly tested and found pitfall we might want to fix while it's hot:

From the host:
```
$ scp root@cam:/tmp/done .
sh: scp: not found
```

On camera (to make scp in $PATH):
```
# ln -s /home/yi-hack-v3/bin/dropbearmulti /usr/bin/scp
```

From the host:
```
$ scp root@cam:/tmp/done .
scp: can't load library 'libutil.so.0'
```

On camera (to make scp in $PATH):
```
# ln -s /home/yi-hack-v3/lib/libutil.so.0 /lib
# ln -s /home/yi-hack-v3/lib/libcrypt.so.0 /lib
```

From the host:
```
$ scp root@cam:/tmp/done .
done
```

Please release with these links or may be consider to move dropbear to rootfs?

shadow-1, 2017-12-24:
@dvv 
This issue has quietly been fixed now. You can try the latest release files.
I never knew of this issue. I tested with WinSCP and it didn't appear to be an issue logging in and browsing.

The issue is due to scp running as a non-interactive shell on the remote device (camera) which does not execute `/etc/profile`. I couldn't find out a nice solution quickly so I just created symlinks

Unfortunately we can't add dropbear to the rootfs partition due to lack of free space. Rootfs essentially has almost no free space. My preference is to keep the little available free space available for additional Busybox utilities as required.

dvv, 2017-12-24:
Thank you!
Right, rootfs is already about full:
```sh
/tmp/sd # ./toybox df -h
Filesystem     Size  Used Avail Use% Mounted on
/dev/root      1.2M  1.1M  128K  90% /
tmpfs          9.0M  4.0K  9.0M   1% /dev
/dev/mtdblock5  13M  8.1M  4.5M  65% /home
tmpfs           16M  824K   15M   6% /tmp
/dev/mmcblk0p1 7.4G  6.8G  682M  92% /tmp/sd
```