shadow-1/yi-hack-v3 #97: Pic resolution ?

nicolasplanchon, 2017-12-18:
Hi there,
I have a little issue: I shot some pics, and randomly some pics haven't the same format. My reference resolution is 1080*1920 and I get some pics in 320*640.

Do you know how I can oblige the camera to take pics with a certain format please ?

Thank's in advance,
Nicolas.

shadow-1, 2017-12-18:
@nicolasplanchon 
Just for my understanding, how are you getting pictures from the camera?

nicolasplanchon, 2017-12-18:
Hello @shadow-1 I use the native app to take the pic :)

shadow-1, 2017-12-18:
@nicolasplanchon 
I am not sure how Xiaomi implement the snapshot feature. However I can confirm that the camera contains two concurrent video streams.

The 'SD' stream in the app is 640 x 320 whilst the 'HD' stream in the app is 1920 x 1080 for those with 1080p cameras or 1280 x 720 for those with 720p cameras.

I believe you may have found a bug in the official Xiaomi firmware!