shadow-1/yi-hack-v3 #98: Yi Home 17CN can't access web interface

xiaoaizhao, 2017-12-20:
Hi,

I'm using Yi Home 17CN and I have downloaded the corresponding firmwares and followed every steps in your instruction. However I'm unable to open the camera web interface by IP address. 

Any idea/suggest of what might be wrong and how to fix it?

Thank you so much for all of your help!

shadow-1, 2017-12-20:
@xiaoaizhao
The most common issue when you can not access the web interface is that the firmware has not managed to get flashed.

A few common questions:
- Are you sure your microSD card is FAT32 formatted?
- When you first flashed the firmware, did you notice a yellow light blink on the camera for approximately 30 seconds the first time it booted up?

---

A few simple things to try out:
- Try flashing an old version of the firmware before flashing the newest again. Pay careful attention when the camera first boots to make sure the yellow light is blinking.
- Use a small capacity microSD card (you 16GB or less). Larger cards default to exFAT filesystem and it's not straightforward to force FAT32 format.
- Use a different microSD card. There is compatibility issues with some cards. I have no idea why.

xiaoaizhao, 2017-12-20:
@shadow-1 
I have tried a few different MicroSD cards of different capacities that are FAT32 formatted.

Here is how it always happen:
After inserting the microSD card and power on, it shows a solid yellow light for 5-10 seconds.
Then it makes a "click" sound which starts rebooting.
Yellow light starts to blink for <5secs and the voice notification starts "Welcome to Xiaoyi, waiting for connection". The yellow light will keep blinking until I connect it to WiFi.

May be the firmware wasn't flashed in? Where can I find some old version of firmware to try?
Thank you!

shadow-1, 2017-12-21:
@xiaoaizhao 
It definitely sounds like the firmware failed to flash for some reason. Like I said, some particular SD cards do not work properly. I have real trouble determining why this is the case. Even when monitoring the console through serial link, I am unable to determine why it does not flash with some cards. There are no meaningful error messages displayed by U-Boot.

I suggest going back to stock firmware or an older version of the hack before going back to the latest version again. I just realised that only the latest version introduced Yi Home 17CN support.

- The hack is available from this page: https://github.com/shadow-1/yi-hack-v3/releases
- Recovery images to go back to stock firmware is available here in the 'yi-hack-v3' folder: https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr