shadow-1/yi-hack-v3 #99: rsync?

mike-s123, 2017-12-23:
I understand it's probably not going to fit in rootfs (amd64 binary is ~400K), but rsync would make things MUCH simpler than ftp.

Since you have the toolchain setup, any chance of compiling rsync?

Also, perhaps add /tmp/sd/bin to PATH, which would avoid having to provide an absolute pathname, and would provide support for future expansion. Or even better, /tmp/sd/yi-hack-v3/bin, and then startup.sh moves to /tmp/sd/yi-hack-v3/startup.sh, etc.

shadow-1, 2017-12-24:
@mike-s123 
I quickly tried cross-compiling rsync. The filesize ended up being ~500kb. The smallest I could get it whilst not removing features which I thought would be very helpful is ~400kb. This makes it too large to include on the camera.

Perhaps with a bit of time I could work something out to include it on one of the camera partitions.

Your suggestion on altering the PATH variable and providing a way to execute custom programs and libraries from the microSD card is excellent. I have meant to do this but keep on forgetting to implement it.

I have squeezed this into the latest release. So whilst I didn't include rsync in the latest release on yi-hack-v3. I have modified the PATH variable along with the LD_LIBRARY_PATH variable to make it easy to execute from the microSD card.

dvv, 2017-12-24:
I believe with #96 done and availability of scp clients such WinSCP etc. `rsync` is not of priority. 
Same considerations imo apply to (pure-)ftpd.

mike-s123, 2017-12-24:
> such WinSCP etc. rsync is not of priority.
You may wish to do some research on the capabilities of each. SCP (or ftp) doesn't provide anything close to what rsync can do. There is no comparison.

dvv, 2017-12-25:
Indeed. `dropbear` package allows for `sshd/ssh` which allows for remote executing of nearly anything including robust copying etc. `rsync` is hardly more than a well-done copier.
Personally, I would like to have both tools, but when it comes to priority, I would personally prefer the first.

letinas, 2019-01-14:
I understand that rsync can't be installed on the camera because of the memory restraints.
How can I compile and put rsync on the memory card?

shadow-1, 2019-01-18:
@letinas 
I would have to look into this at some stage. From memory it was not difficult to cross-compile using the SDK.

You may need to Google a bit to learn how to cross-compile applications for the camera. However I have provided a rough guide on how to produce custom firmware (including some cross-compiling) in this issue: https://github.com/shadow-1/yi-hack-v3/issues/23