shadow-1/yi-hack-v3 #100: Remote update firmware

galadril, 2017-12-24:
It could be a nice option to update the firmware using the webui

shadow-1, 2017-12-25:
@galadril 
I would have to think of a neat way to implement this. At the end of the day, the two firmware files have to be uploaded to the microSD card and the camera rebooted for a firmware update to occur. This is how U-Boot has been configured on these cameras.

The way I currently perform firmware updates is upload the firmware files to the microSD card remotely using FTP and then reboot the camera through the web interface.

CrescendoFang, 2018-01-01:
Tried to update the 17CN camera fw from 0.1.4 beta2 to 0.1.5 failed....
After uploading the 2 files to /tmp/sd and reboot, it just normal boot, the fw stays at 0.1.4...

Is there any procedure to update the camera?

bachya, 2018-01-30:
@shadow-1 What you describe above sounds totally feasible – where exactly do you upload the new firmware files to? `/root`? `/`?

shadow-1, 2018-01-30:
@bachya
The firmware files must be stored on the microSD card. The microSD card is mounted in the following directory; `/tmp/sd`

I guess the main thing to be careful of is to ensure you are transferring the file in 'binary' mode. I have found that this is the default for most FTP clients (but probably not all).

I then simply reboot the camera via web interface for the firmware update to occur.