shadow-1/yi-hack-v3 #101: 27US can't pair with android app

reginastracii, 2017-12-25:
Camera
Yi Home 27US
Base Firmware
1.8.7.0D_201708091510
yi-hack-v3 Firmware
0.1.5

wifi setup goes ok, web server is up 
won't pair with the app - suggest reset

shadow-1, 2017-12-25:
@reginastracii 
Did you disable ProxyChains-ng on the "System Config" page?

You have the US version of the camera which is not region locked. ProxyChains-ng is required if you have the Chinese version of the camera which is region locked.

This exact behaviour occurs if ProxyChains-ng is not configured. However it is not required in your case.

reginastracii, 2017-12-25:
yes,
I have disable it.

also I tried installing older ver. of the app i had and the one from the firmware rollback folder with no luck.

shadow-1, 2017-12-25:
@reginastracii 
Can you confirm that you are using the latest version of Yi Home from the Google Play Store?

The APK file I host is specific for the Chinese version of the camera which is not available in the Google Play Store.

The link to the app for the International version of the camera is: https://play.google.com/store/apps/details?id=com.ants360.yicamera.international

reginastracii, 2017-12-25:
yes, I have tried the current up to date app, also couple of older version and the chinese you host...just in case. 
it is android phone.

shadow-1, 2017-12-25:
@reginastracii 
Try applying the latest official firmware update over the top of my custom firmware. If the update does not occur (this will be because the current firmware on the camera matches the firmware file), you can downgrade to an older version before upgrading again to the latest. The custom firmware will recover from an official Xiaomi firmware update.

I also host the official firmware updates: https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr
The official firmware can also be downloaded from: https://www.yitechnology.com/firmware/index/class/home

reginastracii, 2017-12-25:
ok.
I looked at the logs on the SD card:
I'm pretty sure the problem was on my side with the isp not allowing connections to the chinese servers. 
used a VPN and the paring went ok. 
sorry for wasting time.

shadow-1, 2017-12-25:
@reginastracii 
Wow that is strange! Thanks for letting me know the solution.
If someone has the same issue, I will ask them to try on a different network or use VPN and see if it works.

reginastracii, 2017-12-25:
noticed this in the log file: 

in choose_server, region_id = 17, api_server = https://api.us.xiaoyi.com, sname = familymonitor-y18, dlproto = mius

followed by some failed and crashed logs.  one was arp_test crashed! and some more - I did not keep them and it did write over the logs I guess it's just keeping the last boot ones. 

so I checked that I have no ping ot xiaoyi.com, vpn-ed to different location and it all went smooth after that.

shadow-1, 2017-12-30:
@reginastracii 
Thanks for the information. It may provide with some hints on what to look into.

The Xiaomi applications running on the camera are not perfect and do crash occasionally. There is a watchdog program that runs on the camera. If there is an issue with any of the Xiaomi programs, the crashed program gets restarted or the whole system gets rebooted.