shadow-1/yi-hack-v3 #102: Problem with  freevpn.ninja ?!

vadvik, 2017-12-27:
Hello!
I cant to download list of proxy. Yi Dome 720. China version
[image]

shadow-1, 2017-12-30:
@vadvik 
It appears the https://freevpn.ninja is down. I have checked this with http://www.isitdownrightnow.com to confirm that it is down for everyone.

Hopefully this is just a temporary glitch. However for the time being, a proxy server which is located in Mainland China needs to be found manually. There are many websites with lists of free proxy servers. I have developed the "Test Proxy List" feature which will be very helpful in verifying that the entered proxy servers work as expected.

In the future I may develop the "Download Proxy List" feature to fetch proxy servers from another regularly updated source if I can find another source which hosts proxy servers in a format which can be easily parsed such as TXT or JSON.

LaFed, 2018-01-05:
do we need to update this list from time to time?? I've manually added a few and was wondering if needed to keep the list healthy

CrescendoFang, 2018-01-05:
use this entry for time being, its really working for me....
http://spys.one/free-proxy-list/CN/

shadow-1, 2018-01-09:
@LaFed 
As long as the proxy server is alive, there is no need to update the list. It is good to check whether the proxy servers still work from time to time.

Au8ust, 2018-03-25:
@CrescendoFang  Could you please guide me how to use the entry?

CrescendoFang, 2018-05-11:
@Au8ust just take those IP and port and paste it below [ProxyList] with format like this:

socks5 223.241.214.44 1217
http 120.79.78.178:1080

socks5 for socks5 proxy
http for http and so on....

the variations are: http, https, socks4, socks5

omit the : between IP and port....

save & test

closed 2020-10-05 by vadvik