shadow-1/yi-hack-v3 #103: Firmware upgrade

vaidyasr, 2017-12-28:
Can I upgrade the firmware from 2.1.0.0A_201703071456 to 2.1.0.0C_201710091352? I have currently installed 0.1.4-beta2

shadow-1, 2017-12-30:
@vaidyasr 
With this version of yi-hack-v3 (or newer), it is safe to perform Xiaomi firmware updates.

CrescendoFang, 2018-01-06:
Yi 17CN upgrade from 1.8.7.0C_201705091058 to 1.8.7.0D using 0.1.5 succeed, however the version remains the same 1.8.7.0C but the camera says up to date...

Strange....

shadow-1, 2018-01-09:
@CrescendoFang 
Thanks for letting me know. I will have to download the new firmware, unpack it and see how it has changed from the previously released firmware.

CrescendoFang, 2018-05-23:
@shadow-1 any news for this project?

batrarobin, 2018-07-31:
Yes, I just updated it and it still works.

closed 2019-05-20 by vaidyasr