shadow-1/yi-hack-v3 #104: Yi Camera with MI Home App

vaidyasr, 2017-12-28:
I recently bought Xiaomi Xiaofang camera and it works fine with the Mi-Home app from playstore. The Mi-Home app also detects my existing 2 Xiaomi Yi cameras automatically and added them. But it doesn't detected the Xiaomi yi 1080p with yi-hack-v3. How to make it work with Mi-Home?.

ainen, 2017-12-28:
Have you gone through the [Step by Step guide](https://github.com/shadow-1/yi-hack-v3#getting-started---step-by-step-guide) entirely? Step 10 and 11 are very important to the whole process.

vaidyasr, 2017-12-28:
I can able to pair with YI home app, but am looking for pairing with Mi home app.

shadow-1, 2017-12-30:
@vaidyasr 
These mobile apps are developed by Xiaomi. I do not have any input into them.

I have heard the Mi Home app is quite convenient. However it is designed for the Mijia range of devices. I believe Xiaomi developed a plugin for the original Yi Home camera and it works with the Mi Home app. However Xiaomi have not developed plugins for the other Yi cameras at this stage to make them compatible with the Mi Home app.

yaffadev, 2018-03-13:
I have the same problem.  It turns out that only the Chinese version Yi camera will pair with Mi app. I am thinking of updating the firmware to the Chinese version. Not sure if it works or not though.

closed 2019-05-20 by vaidyasr