shadow-1/yi-hack-v3 #105: Why the video and feed not routed through proxy

mendellev82, 2017-12-29:
According to your statement "Performance is not degraded as the cameras video/audio feed is not routed through the proxy server." Since if we using the proxy, the Yi server there should be the proxy IP isn't it?
If in that case the video/audio feed should be routed through proxy isn't it? Can you explain a bit why video/audio feed is not routed through the proxy server, how it work?

shadow-1, 2017-12-30:
@mendellev82 
I have no problems in explaining this in a little more detail.
I do not know exactly every detail on how these cameras work. However I have a reasonable idea.

In essence, there are two types of connections occurring on the camera, HTTP and P2P. The camera authenticates and some commands get executed via the HTTP connection whilst all video/audio data goes through the P2P/cloud service. I can't remember exactly but from memory commands executed through the Yi Home app go through the P2P connection (commands such as turn on/off power LED, perform PTZ actions on dome cameras etc).

What we are really interested in hacking is the camera authentication process. Once the camera is authenticated, it is usable within the Yi Home app. The Xiaomi server checks whether the camera resides within Mainland China as part of the authentication process (only for region locked cameras). This check definitely occurs via HTTP connection to the Xiaomi server.

This hack intercepts all HTTP requests and re-routes them via the proxy server. As video/audio data goes through the P2P connection, it is never routed via a proxy server.

mendellev82, 2017-12-30:
@shadow-1 
Thanks so much. Your explanation is awesome. It clear my doubts.