shadow-1/yi-hack-v3 #112: Will you support 48CN?

Aurora66, 2018-01-05:
I recently purchased this Yi Home 1080p
the QR code is: 48CNY31NQV1AHX170518
Looks like this device is not on the support list.
Do you have plan to support it soon? 
Thank you.

LaFed, 2018-01-05:
I believe all Yi 1080p Home is supported...

shadow-1, 2018-01-05:
@Aurora66
I am not aware if the 48CN version of the camera is any different to the previous versions.

The only camera I had to be careful with so far is with the Yi Home 720p. Xiaomi changed the wireless chipset with one of the versions which broke this firmware till I included both wireless drivers in the one firmware.

Aurora66, 2018-01-05:
@shadow-1 Thank you very much for replying to me. The reason why I raised this issue is that my Yi Home 1080p always fails at the last step. It complains something like "cannot pairing. wrong device selection." in Mandarin.

@LaFed Which firmware did you download for your 48CN? I tried 

rootfs_y18 | home_y18

rootfs_y20 | home_y20 | F

but neither works. Both failed at the last step with "wrong device" message. 

Do you mind sharing some of your experience? 

Thank you very much!

LaFed, 2018-01-06:
@Aurora66 use the y20 ones... Just know that Yi 1080p Home is all the same... ignore the serial for now. Im not sure which last step u are talking... just follow the instructions correctly and you'll be fine... You wont be able to pair them unless completed the proxychain step... And take note on the app that you are downloading on your phone, it matters

elaprty, 2018-01-22:
im having the same issue, it connects to the internet, I also checked the config set up and its working. but the pairing with the app is not working. even though Ive downloaded from the china app store.

tcmean, 2018-02-08:
For those that are having issue with pairing with WiFi properly set up, please download the app from box.com drive instead of Mi Appstore. It seems the latest app version is blocking the pairing

RaymondCKL, 2018-03-22:
when connect to /cgi-bin/read_proxy_config.sh. It only shows 404. Can anyone help? Thanks.
my device is 48CNYPVIZU