shadow-1/yi-hack-v3 #115: YI Home 22CN

myeyepl, 2018-01-06:
Please let me know if you will work on a firmware for 22CN version.

shadow-1, 2018-01-09:
@myeyepl 
Support for the original Yi Home cameras (prior to 17CN and 27US) is still a work in progress. Have a look at this issue: https://github.com/shadow-1/yi-hack-v3/issues/25

ezar, 2018-01-21:
+1 please!!!!

sabat54i, 2019-02-14:
Hi,
Any news about 22CN ? It doesn't work for me 
Thank you

shadow-1, 2019-02-20:
@sabat54i @Corwin616 
I am too busy at the moment, however I will try to revive this portion of the project as I think it is somewhat working as is.

The biggest trouble is that I do not have this camera at all, so it is challenging to sort out issues. It is based on an older chipset than the cameras which are currently supported.

The other cameras which are supported are all based on a newer chipset and are all very similar. So if it works on one, it appears to work on all.