shadow-1/yi-hack-v3 #116: How can I compile curl for the camera?

josemonino, 2018-01-07:
I have seen in #79 that it is possible to have curl working in this camera.

I downloaded the lib https://github.com/utya1988/yi-hack/raw/master/sd/test/curl/libusr/libz.so.1.2.8 and renamed it to libz.so.1 but I don't know how to compile curl for Yi Home 27US.

Thanks

frekel, 2018-01-08:
For future reference:
```
cd /home/yi-hack-v3/sbin/
wget https://github.com/utya1988/yi-hack/raw/master/sd/test/curl/curl
chmod +x /home/yi-hack-v3/sbin/curl
cd /home/yi-hack-v3/lib
wget https://github.com/utya1988/yi-hack/raw/master/sd/test/curl/libusr/libz.so.1.2.8 -O libz.so.1 
```

shadow-1, 2018-01-09:
For those that wish to compile `curl` from source. Keep in mind that I have tried to disable many features which I thought would not be very useful on the camera in an effort to keep the file size down (to conserve free space). 

In addition, keep in mind that these instructions build `curl` statically. This means that the `curl` executable is larger, however there is no dependency on external dynamic libraries.

These commands also assume you have successfully installed the cross-compiler which is included in the SDK.

```
git clone https://github.com/curl/curl
cd curl
./buildconf
./configure --host=arm-hisiv300-linux --prefix=/home/yi-hack-v3 --disable-shared --disable-libcurl-option --disable-dict --disable-tftp --disable-pop3 --disable-imap --disable-gopher --disable-ipv6 --disable-manual
make
export DESTDIR="$PWD/_install/" && make install-strip
```
Once these commands are executed successfully, copy `./_install/home/yi-hack-v3/bin/curl` to `/home/yi-hack-v3/bin` directory on the camera. Depending how the executable file is transferred, you may need to change permissions of the executable with the following command; `chmod +x /home/yi-hack-v3/bin/curl`.

josemonino, 2018-01-25:
@shadow-1 @frekel  THANKS!!

closed 2018-01-25 by josemonino