shadow-1/yi-hack-v3 #118: Manual network setup

Xyrax, 2018-01-08:
Hi, 

first, thanks for all the hard work and effort you put into this project! 

Is it possible to set the network settings manual? Maybe with a start script of some sort? Because my camera needs glasses or reading lessons, it just won't read the QR code and even when it does (kinda random) it says wifi password incorrect which is also incorrect...

So before I'm returning the cam, is there still any hope to get this thing working?

frekel, 2018-01-08:
Hi,
Are you sure you have the correct app?  I had the same thing when I tried connecting with the MiHome app. You really need the Yi Home app.... one char does matter ;)

Greets,
Frank

shadow-1, 2018-01-09:
@Xyrax 
Another thing to consider is if you have the correct Yi Home app. Remember that Xiaomi have a separate app for their International cameras and Chinese cameras.

From memory, the same issue occurs if you use the Yi Home International app with a Chinese Yi camera (most likely vice versa as well).

Xyrax, 2018-01-09:
Thanks for your help!

My camera is an international version (15USY2U***) and I'm using the Yi Home App from the Google Playstore.

I changed my WIFI name to a one without special characters, also my wifi password, but it still won't read the QR code right now. Is there a limitation containing the length of the password? Mine has 63 characters, that's maybe to much? 

I also got some logs, if that helps...

shadow-1, 2018-01-09:
@Xyrax 
Ah that reminds me! Yes there is definitely a limitation on the maximum WiFi password length. I have come across this issue before when I tried using the camera on a WiFi network with a 64 character password.

Unfortunately the only way to get it to work is to use a shorter password. I would have to dig into the firmware to determine the maximum WiFi password length.

Xyrax, 2018-01-09:
@shadow-1 
Considering the logs from the camera, my best guess would be around 45 chars, at least that's the point where the logs went haywire :D 

`[./rmm][1/19/6:51:17:912]: rmm got /tmp/got_wpa ssid(No More Mr Wi-FI) pwd(dTZZmlE8tOQBszpgCiHxApFJB89EyNgKlRrbBHddKXDfZgg¦ÉÈ†ÇË†Äº»Å:òˆÅÏ±œ‘””“ËÆ´x¥›GùúóŠË¸ÈÉ±µ°²Æ–ÆÏ˜—˜È¦ÈÌ¸ÈÇ‹ÉÇÇÆÆ†ÈÆ·¦È˜È†ÇÈ†Æº»ËÊºˆÌÏ°Ï•””“ÇÆµ¹¬•Ç·ª—ŠÊÈÈÉ±µ°²Æ–ÆÏ˜—˜È¦ÈÇ¸ÈÇ‹ÉÇÇÆÆ†ÈÆ·¦È˜È†ÇÎ>*8<3*EPt_2=6s+	")P[j5Uz<R\sa}m^/ÿþoÿÿÿüÿýÿÿñ·ÿöÿþ¬ûÿÿÿóÿþ>TÏô±¯ÿ›ÿÿþÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿôÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿù¶) bind(EUez587LlkTt7qmO)`

No More Mr Wi-FI was the old name of my WIFI, and dTZZmlE8tOQBszpgCiHxApFJB89EyNgKlRrbBHddKXDfZgg******************** my password, but after the "g" it seem like an overflow or sort like that?

shadow-1, 2018-01-10:
@Xyrax 
Couldn't work it out quickly from the firmware files. My assumption is that when the WiFi password gets too long, the end portion of the password gets gobbled up by other data that is stored in the same area.

Xyrax, 2018-01-10:
@shadow-1 
So I changed my password to a shorter one, but the camera won't read the QR code. I'm desperately trying, but it just won't work... is there anything I can do to get the camera read that code?

shadow-1, 2018-01-10:
@Xyrax
Can you confirm that you are using the correct smartphone app.

Please post what version of Yi Home you are currently using.

Xyrax, 2018-01-10:
@shadow-1 
I'm using Yi Home Kamera Version: 2.20.18.0:20171218 the latest release I guess?

Xyrax, 2018-01-11:
@shadow-1 
Just to let you know Yi/Xiaomi confirmed there is something wrong with the camera. I flashed the cam back to the original firmware but it didn't solve the problem. I attached the logfile how they figured out that the cam is broken for your interest.
I will report back when they replaced the camera :) 

[log.txt](https://github.com/shadow-1/yi-hack-v3/files/1622249/log.txt)

shadow-1, 2018-01-11:
@Xyrax
Wow. I never realised some of these cameras were defective. Thank you for sharing the log. I don't exactly understand what hints the log contains that there is an issue with the camera. However it might help in the future.

Xyrax, 2018-01-12:
@shadow-1 
Mybe it isn't about what is in the log file, maybe it is more about what isn't? Kinda philosophical ;-)

shadow-1, 2018-01-12:
@Xyrax 
True true

iorpass2, 2018-01-13:
I would also like to know if I can access wpa_supplicant.conf to manually change the WiFi settings. I have a long passworrd with special characters, and I have not been able to connect with this password (the camera never recognizes the QR code). 

I am on the 27US Yi Home camera. Is there any way to manually edit the WiFi settings?

maaaatteo, 2020-06-01:
> @iorpass2
> There are a few different ways of achieving what you want.
> 
> `wpa_supplicant.conf` is located in the following directory; `/tmp/wpa_supplicant.conf`.
> Unfortunately it gets generated every time the camera boots.
> However one potential solution is to modify `wpa_supplicant.conf` after the camera boots.
> 
> The easiest way to achieve this is probably created a folder called `yi-hack-v3` on the microSD card can create a shell script named `startup.sh`. A shell script should be enough to modify `wpa_supplicant.conf`.
> 
> Keep in mind that I have not written the shell script. Also keep in mind that I have not tested this method to confirm that it works as expected.

is there any update for this subject?