shadow-1/yi-hack-v3 #119: Time Stamp

Izzybuzy, 2018-01-09:
I have 27US Yi Home Camera.

Time stamp appeared on the top middle of screen when view under Yi Home App.
But after using the sd card in to view in pc, there wasn't any time stamp at all. Is this normal?

Also, under date modified, it always showed 5 hour ahead. But I already set camera time zone to Eastern time UTC -5 in the app.

shadow-1, 2018-01-10:
@Izzybuzy 
Can you post a couple of screenshots describing your situation. I know some cameras burn a timestamp into the video files (for example Yi 1080p Dome) whilst some do not (for example Yi Dome).

With regards to the modified time being incorrect. The timezone setting is set by the smartphone app and gets set for the Xiaomi programs running on the camera only. I believe the actual system system timezone is never altered, hence why modified date/time are off. It is possible to alter the system timezone but it may or may not interfere with the Xiaomi programs running on the camera.

Izzybuzy, 2018-01-10:
Top screenshot is play direct from SD card using FTP. I even took the sd card out and copy file to pc and still no timestamp. I only see a YI logo burned in on lower lefthand corner and that is it.

Bottom screenshot is taken when viewing from within the Yi Home App. As you can see, timestamp 01/10 12:45PM is clearly visible.

[[image]
[image]

If a YI Logo can be burned in and you said Yi 1080p Dome have time burned in, if it isn't too much trouble, would it be possible for you to add a timestamp burned in on your next firmware update for Yi Home 27US?

For the modified time, I included a screenshot showing the modified time issue. the Motion Alert file **45M36S.mp4** should be 12.45:00PM instead of 5:46:00PM. Why would they have a timezone setting within the app, but it doesn't help altered to save file in correct timezone. I tried setting to other timezone and it is still 5 hours off. Assuming, Yi app worked and sent time to Camera correctly, then firmware is ignoring the correct timezone. then firmware need to be fixed. Of course, I will check with Yi, it is also possible that their app doesn't send timezone to camera. what do you think? 

[image]

BTW, would you be able to add auto upload feature to Drop box and Google Drive on WEB GUI in future firmware? In particular, it will let user set specific time of day when it will upload and if possible camera will know only to upload new files rather than upload everything within sd card?

Thank you.

](url)

Izzybuzy, 2018-01-10:
FYI, I did some research

shadow-1, 2018-01-11:
@Izzybuzy
I actually have not modified the video feed that Xiaomi developed. I'm not sure whether I will be able to in the future either.

From your screenshots, it is clear that the video doesn't have the date/time burnt into the video. In the app, the date/time is overlayed.

On the Yi 1080p Dome when I did have one for a little while for testing, the date/time is burnt into the video feed on the top left hand corner. In addition in the app, there is the date/time overlay just like in your screenshot. You actually see two sets of date/time in the app.

I am aware of the issue where modifying the time zone shifts the time in the app. Again I believe this is a limitation in the Xiaomi app and I dont think I will be able to fix it. However the file modified time should be fixed (from memory).

With regards to auto upload to a cloud service feature. I believe there are people who have developed scripts that achieve this. If you can link me with a robust script, I might be able to include it in the next firmware release.

Izzybuzy, 2018-01-15:
You said "However the file modified time should be fixed (from memory)." So you fixed this on firmware before and it still happening due to Xiaomi app, right?

Anyway, I found an auto upload Google Drive script from another github project  for xiaomi smart ip camera.
https://github.com/porunov/xiaomi_gdrive
I am not sure if this is compatible to YI Home 27US that I have or any other YI cameras for that matter. If it compatible, I hope you can apply this to your next firmware update.  Please see if you can integrated into WEB GUI, so user can input and modify setting with ease.

Thanks.

DaveBettz, 2019-11-26:
Did you ever manage to include the timestamp?

Alex2x3x, 2020-01-30:
I have the same problem no timestamp, don't why it can't be applied as other dome cams have them.  I had 2 previous dome cams with time stamps but  the sound quality news rubbish.