shadow-1/yi-hack-v3 #120: FTP issue ?

nicolasplanchon, 2018-01-09:
Hi everyone,
Juste a litte problem trying to reach my cam through FTP, let's see:
[image]

FTP Server enabled of course.

Any idea please ? :)

Best regards,
Nicolas.

shadow-1, 2018-01-10:
@nicolasplanchon 
Everything appears to be working properly. The default directory when connecting via FTP is `/root`. This is the 'home' directory (or 'my documents' folder in Windows) for the user root.  By default this directory is empty, hence why it looks empty. Try navigating up one level to `/`. You should then see the rootfs data.

CrescendoFang, 2018-01-12:
don't forget to set the tile transfer mode to BINARY