shadow-1/yi-hack-v3 #122: Pairing timedout

ronendayan, 2018-01-10:
Hi
I got the YI Dome
I put the files you wrote on the SD card (without the YI firmware file)
And now i can't access the camera
the http url doesn't work.
I try to reset the camera and put all 3 files on the SD card but without success,
Please advise what to do next

ronendayan, 2018-01-10:
So i got the recovery image from here:
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr
and the camera back to normal.
I will try to flash again the hack version
Thanks

shadow-1, 2018-01-10:
@ronendayan 
Great that you managed to get to the Recovery image. This will get you back to stock firmware. Just try again and see if it works properly now.

ronendayan, 2018-01-11:
Tested it again and its working fine,
Thanks

closed 2018-01-11 by ronendayan