shadow-1/yi-hack-v3 #123: Proxy issue

moonboxer, 2018-01-10:
Hi,

Recently I bought a Yi 1080p Home camera but I encounter some difficulties with the Proxy setting.
First , I couldn't download any proxies which is fine for me, so I tried to manually set it up. Then after trying almost 200 CN proxies, it appears that not a single one is workable. Do you guys have any idea about that?

shadow-1, 2018-01-10:
@moonboxer 
Just need to try proxy servers from a different website which is updated a bit more frequently. For example: http://spys.one/proxys/CN/

I also suggest sticking with Socks4 or Socks5 proxy servers as they are more likely to work properly.

moonboxer, 2018-01-10:
Hi shadow,

Thanks for the suggestion, I will try tonight!

Best Regards,

mrzang1512, 2018-01-13:
how to config proxy at website http://spys.one/proxys/CN/ to ProxyChains-ng Configuration

moonboxer, 2018-01-13:
Hi

After the proxy list add about 10 socks4 or 5 proxies like below:

socks4 139.199.230.242 1080
socks4 138.199.230.242 1080
socks4 136.199.230.242 1080
socks4 135.199.230.242 1080
socks4 193.199.230.242 1080
socks4 134.199.230.242 1080

mrzang1512, 2018-01-13:
i added but when after test , it show : ERROR: Could not connect to api.xiaoyi.com, ensure that the configured proxy servers are operational. Could you teamviewer to my computer , support me.

moonboxer, 2018-01-13:
Just try more proxies listed from that webside. Only a few works but still
you have a higher chance to find one working than other proxy website.

mrzang1512, 2018-01-13:
tks you ,i'll try

gm1231s, 2018-01-15:
@mrzang1512
Did you manage to connect to any of the servers in the list provided by moonboxer?

I tried the list and I could not connect to any one of them.

@moonboxer 
Are you still able to connect to the proxy servers today?

[image]

[image]

mrzang1512, 2018-01-16:
@gm1231s you try proxy CN only , update at http://spys.one/proxys/CN/
Ex 
socks5 122.4.161.117  1080
socks5 183.158.142.54  1080
socks5 183.6.168.196  1080
socks5 221.209.62.170  1080
socks5 61.145.213.14: 1080
socks5 117.135.132.107  3000
socks5 125.79.199.78  8888
socks5 120.133.6.6  9065
socks5 121.196.207.36: 1080
socks5 124.163.236.110  1080

moonboxer, 2018-01-16:
@gm1231s <https://github.com/gm1231s>  It is connected once for all. You
don't need to re-configure it once to successfully connect to a proxy.

gm1231s, 2018-01-16:
@mrzang1512
Thank you, I finally managed to connect.

NeoDeGenero, 2018-02-18:
Hmm, so the initial list I had was working for about 3 weeks. Until about yesterday when all the sudden I get the audio message saying that the Camera cannot be used outside of China. I've updated the list, and now seems to be working. Will update if there's any changes.

Ahmadbe, 2018-03-19:
Hello,
I have a problem connecting to my mobile.
when I try to download proxy it show me:

wget -O /tmp/temp_proxy_list.txt https://freevpn.ninja/free-proxy/txt 

wget: bad address 'freevpn.ninja'

chilskater, 2019-06-02:
https://github.com/TheCrypt0/yi-hack-v4/releases/tag/v0.2.0 is this the same camera?