shadow-1/yi-hack-v3 #124: Creating Linux images from scratch

diegosueiro, 2018-01-10:
@shadow-1,

Today I found this [repository](https://github.com/hi35xx/hi35xx-buildroot) and I'm wondering if you already tried this. It should be interesting if we have the possibility of booting an alternate Linux image (kernel and rootfs) from the micro sdcard.
I'm especially interested in getting this working for the Yi Dome 1080p and I will invest some time on this. But I'm still not sure how your current hack works. If you could give details on what happens after the kernel boots it will help me a lot.

Another thing is, do you have details of the Dome 1080p hardware? Which camera sensor is, flash and ram size and other aspects? Do you know if it is possible to have access to the SoC UART console?

As a side note. I read that you are currently trying to get rtsp working (which is wonderful), have you already saw this [project](https://github.com/hi35xx/live-streamer)?

FunkyM, 2018-01-25:
Cool find. This appears to be a fork of the https://buildroot.org/ project with injected hisilicon board configuration and some SDK files for audio/video support. The goal appears to be to compile a "live-streamer" RTSP application for the camera but it doesn't appear complete.

I managed to add the missing pieces and a custom toolchain for the Yi 1080p Dome and everything "buildroot" compiles appears to work fine on the camera. It appears I might even already have proper jffs2 images being generated for the camera, too.

I am going to test the video functionality and to get "live-streamer" running properly before attempting to flash the generated images but it looks very promising since the binaries work nicely.

diegosueiro, 2018-01-26:
@FunkyM,

Yes, it is based on Buildroot.

Thanks for your report. 
Do you know if the original image has support for kexec (both Kernel and userspace tool)?
If so, it is easy to get it booting an alternate kernel and rootfs from micro sd and no modifications will be needed at bootloader version.

If the live-stream project that I linked works fine I think it's a good candidate to bring rtsp server into the hack.