shadow-1/yi-hack-v3 #127: Dome Camera 720p solid yellow after flash

Stuggy, 2018-01-13:
I installed the latest download as of Jan 13, 2018.  After the dome boots up it rotates to the furthest position and then keeps making the motor noise.  After restarting it has already rotated and just buzzes while it appears to be trying to turn more?  The light is solid yellow.

ronendayan, 2018-01-13:
I had some trouble with it my YI Dome also,
try to flash it again, that hardware has a problem with those SD cards so it can crash / fail

shadow-1, 2018-01-28:
@Stuggy 
@ronendayan has the right idea. Just re-flash the firmware to recover from an issue that stops the camera from booting. Often, you will need to downgrade to an older version before upgrading again to the latest version.

Stuggy, 2018-01-28:
Thanks for the replies.  I am still using the downgraded stock image but will try again.