shadow-1/yi-hack-v3 #129: Pure-FTPd doesn't work with FFmpeg

bachya, 2018-01-14:
I know I was the one to originally suggest Pure-FTPd... 😄 

My original use case is to use FFmpeg to render videos from the SD card via commands that look like this:

```
ffmpeg -i ftp://root:password@192.168.1.121:21/tmp/sd/record/2018Y01M14D01H/55M00S.mp4
```

...which, with the latest firmware version, gives a `moov atom not found` error. Info is hard to come by, but it seems like [Pure-FTPd may produce content that FFmpeg can't parse](https://ffmpeg.org/pipermail/ffmpeg-devel/2013-July/146167.html).

No clue how you can help this; you've done the right thing. Wondering if there'd be some way to toggle between Pure-FTPd and ftpd?

shadow-1, 2018-01-28:
@bachya 
Yes, switching between Pure-FTPd and ftpd is possible. For the time being, you can modify `/home/yi-hack-v3/script/system.sh`.

bachya, 2018-01-30:
@shadow-1 Awesome, worked great – thank you.

Until a more official mechanism is put in place, for anyone stumbling on this:

1. Open up `/home/yi-hack-v3/script/system.sh` in a text editor (via FTP, SSH, etc.).
2. Comment out line 36 (i.e., `pure-ftpd -B`).
3. One line above or below, add in the original command: `tcpsvd -vE 0.0.0.0 21 ftpd -w &`
4. Reboot the camera.

bachya, 2018-02-11:
@ilshatms Not quite; it should look like this:

```bash
if [[ $(get_config FTPD) == "yes" ]] ; then
        tcpsvd -vE 0.0.0.0 21 ftpd -w &
fi
```

nayneyT, 2018-03-03:
Hi guys.

I get invalid directory listing.

Any idea what is wrong and why I cannot connect after making the change?

Thanks.

DeanNotDin, 2018-06-18:
@vpomax I'm also getting this issue!
Running YiCam on 27US with YiHack 3 0.1.5, and HA Running 0.71.0
Getting `FFmpeg fails program arguments must be a bytes or text string, not bool`

DeanNotDin, 2018-06-22:
I managed to get the camera running!
Configuration as follows:
`# YiCamera
camera:
  - platform: yi
    name: Front Door
    host: 10.255.48.12
    password: !secret yicam
    ffmpeg_bin: /usr/local/bin/ffmpeg
    ffmpeg_arguments: '-vf scale=800:450'`

Now I face a different issue! the frame rate is just awful, 1 frame every ~3 minutes.
I tried both pure-ftpd and  tcpsvd, both providing the same results.

willsy555, 2018-07-17:
That's correct. It updates via FTP every few minutes. The live feed option is wip apparently! Maybe!

uli6, 2018-08-22:
@DeanNotDin , same issue here.

It tooks an eternity to update and sometimes freezes. :/

Aiace88, 2018-12-03:
hi. this is my configuration of the cam. 

[image]

but in ha i have the sam error, no display image and in the logs...

2018-12-03 23:42:36 ERROR (MainThread) [homeassistant.components.camera.yi] Error while fetching video: Waiting for ('2xx',) but got 550 [' Error']

someone can help me? thanks

bachya, 2018-12-03:
HTTP 550 means "Permission Denied", so I would guess that the username and password you're using to access the camera aren't valid.

Aiace88, 2018-12-04:
> HTTP 550 means "Permission Denied", so I would guess that the username and password you're using to access the camera aren't valid.

thanks for the info.

the password is correct, is the same i use for ssh e telnet.

this is my configuration in the file configuration.yaml...is correct?

camera:
  - platform: yi
    name: CAMERA-INGRESSO
    host: '172.0.0.x'
    password: password
    ffmpeg_arguments: '-vf scale=800:450'

bachya, 2018-12-04:
Hmm, that does look correct. Can you FTP directly into the camera using those credentials?

azzis77, 2018-12-10:
same issue...

Nixellion, 2019-01-08:
So, i've changed the ftpd line in config, rebooted camera (from webui), changed password and here's what I get now:

`ftp://root:pwd@192.168.1.145:21/tmp/sd/record/tmp.mp4.tmp: Invalid data found when processing input`

OR

```
[mov,mp4,m4a,3gp,3g2,mj2 @ 00000255a213a040] Format mov,mp4,m4a,3gp,3g2,mj2 detected only with low score of 1, misdetection possible!
[mov,mp4,m4a,3gp,3g2,mj2 @ 00000255a213a040] moov atom not found
ftp://root:pwd@192.168.1.145:21/tmp/sd/record/2019Y01M08D12H/00M00S.mp4: Invalid data found when processing input
```

Evanlec, 2019-01-12:
The super long delay caused by using this method with HA makes it nearly useless from my point of view. I don't want to see video from X minutes ago, I want to see live video, that's the only reason to have the camera showing in HA dashboard.

Nixellion, 2019-01-13:
@Evanlec You're right. It is not HA's issue though, it's this firmware's issue. You can't really read unclosed mp4 h264 stream, and camera only saves it once a minute. And there's no 'current frame' file there, not that I could find. If this firmware at least implemented that, saving current frame into a file at least at around 5 fps that would already be a huge achievement. 

That's why I'm ordering another wifi camera. And actually considering building one of my own on Raspberry Pi Zero W and even adding some temp, humidity and light sensors to it to make it the smartest and yet cheapest baby monitor lol. Nothing like that on the market really. 

As for this - well, even 1 image a minute is still better than nothing, and may find it's use in HA or otherwise. Could at least be used for timelapses :D

ajdezigns, 2019-01-18:
> @shadow-1 Awesome, worked great – thank you.
> 
> Until a more official mechanism is put into the UI, for anyone stumbling on this:
> 
> 1. Open up `/home/yi-hack-v3/script/system.sh` in a text editor (via FTP, SSH, etc.).
> 2. Comment out line 36 (i.e., `pure-ftpd -B`).
> 3. One line above or below, add in the original command: `tcpsvd -vE 0.0.0.0 21 ftpd -w &`
> 4. Reboot the camera.

Where and how do I access this file? I've FTP'ed my camera's and I've tried accessing them through telnet but I'm unable to locate the files.

shadow-1, 2019-01-18:
@ajdezigns 
This file is located in the following directory; `/home/yi-hack-v3/script`
It has to be here, otherwise you will not have access to Telnet, SSH, FTP, Web Server etc.

If using Telnet, you will need to edit the file using `vi`.
Alternatively you can transfer the file to your local PC using FTP, edit it locally and transfer the file back to the camera (replacing the file) using FTP.

ajdezigns, 2019-01-19:
I've made the changes to one Yi 1080p Home Cam and two Yi 1080p Dome Cam's but I'm getting the following errors in home assistant 

Log Details (ERROR)
Fri Jan 18 2019 02:15:00 GMT-0500 (Eastern Standard Time)
Error while fetching video: Waiting for ('2xx',) but got 550 [' Error']

I'm unsure how to get it to work.

I've added my Code from my config below as you can see I've tried different ways on each cam but all provide the same error.

`  - platform: yi`
`    name: Kitchen`
`    host: '192.168.0.29'`
`    password: !secret kitchen_cam
`
 ` - platform: yi`
`       name: Entrance`
`    host: '192.168.0.32'`
`    password: !secret Entrance_cam`
`    ffmpeg_arguments: '-vf scale=800:450'
`
`  - platform: yi`
`    name: Living Room`
`    host: '192.168.0.27'`
`    password: !secret livingroom_cam`
`    path: /home/camera/feed`
`    ffmpeg_arguments: '-vf scale=800:450'
`
not sure why it's not propperly indenting here it is in my file though.

ajdezigns, 2019-01-19:
> I have 2 yi dome 720 and sometimes I get the same error but only for a camera

What does your settings look like?

marcogazzola, 2019-01-19:
This Is my config:

- platform: yi
  name: sala
  host: '192.168.1.169'
  ffmpeg_arguments: '-pred 1 -vf scale=800:450'
  password: !secret yi_password
- platform: yi
  name: cucina
  host: '192.168.1.160'
  ffmpeg_arguments: '-pred 1 -vf scale=800:450'
  password: !secret yi_password

The second One Is the camera with problem

ajdezigns, 2019-01-26:
Do you have a sd card in either or cam?

papperone, 2019-03-19:
any clue how to fix this? it's so frustrating...
I keep getting the same error as other user --> Error while fetching video: Waiting for ('2xx',) but got 550 [' Error']
I edit the script btu no changes, but when I FTP to the camera I cannot find the "record" folder, basically "/tmp/sd" is empty, how can I fix this? looks like my camera is not creating any stream file on the SD card :(

Rastablastaman, 2019-05-10:
Hi Aiace88, 
  have you solved your prblem with your cam? is it possbile to contact you and speak italian? i can't see my yi 1080 p cam in home assistan, and i am wonderif if you can help me to set it up!

Thank you very much

> thanks for the info.
>
> the password is correct, is the same i use for ssh e telnet.
>
> this is my configuration in the file configuration.yaml...is correct?
>
> camera:
>
> * platform: yi
>   name: [...]

Rastablastaman, 2019-05-10:
@marcogazzola
Hi Marco, 
  i have an YI 1080 p cam, can you help me to set it up on home assistant? i have any trouble to get it! is it possible to contact you and speak italian?

Thank you very much!

IlTeo285, 2019-05-25:
Hi everybody,

I've changed `system.sh` as suggested.
After rebooting the camera I can't connect anymore. 
Getting the error `Connection refused` using both ssh and telnet

Any ideas to solve that?

thanks

closed 2019-07-21 by bachya