shadow-1/yi-hack-v3 #131: yi 1080p would not connect via proxy

elaprty, 2018-01-19:
Hi,

Help, Im a noob and I have done everything and worked out everything until it reaches the point where I need to Download proxy. It has an error. and I cant work around it.

[image]

Please help!

I have 2 of these and I really need them to work. :(

Thanks!
Elaprty

tiong0526, 2018-01-21:
Try this...
[ProxyList]
socks5 122.192.32.79 7280

elaprty, 2018-01-22:
Thanks tiong0526!! that worked. Now the problem is that the app sees the sd card as broken, and wants to format it, if I do that, the camera stops working. If I dont format It the camera will not save any recording. What to do in this case? I am trying to connect via ftp to see if theres a step I missed in the instructions:

> Using the Camera
> SSH Server
> The SSH server is on port 22.
> 
> Default user is root. Password is blank.
> 
> Telnet Server
> The telnet server is on port 23.
> 
> Default user is root. Password is blank.
> 
> FTP Server
> The FTP server is on port 21.
> 
> Default user is root. Password is blank.
> 
> Startup Shell Script
> On the microSD card. The following shell script is executed after the camera has booted up within a folder named yi-hack-v3:
> 
> startup.sh
> External Programs and Libraries
> Xiaomi cameras have very limited free space available for additional programs or libraries. However these can be stored on the microSD card where they can be utilised from there.
> 
> Place additional programs compiled for the camera in either of these folders on the microSD card:
> 
> yi-hack-v3\bin
> yi-hack-v3\sbin
> Place additional libraries compiled for the camera in the following folder on the microSD card:
> 
> yi-hack-v3\lib

But I dont see any instructions there on what to do with the sd card so that the camera will store recordings in the sd card. Im lost.

My Camera is 48CN. 1080p Ver2

I apologize, I am leeching off you guys, I am not a programmer, but I like tweaking and finding work arounds for things that otherwise wont work. Please help?

closed 2018-01-24 by elaprty