shadow-1/yi-hack-v3 #132: Connect Yi Dome to qnap

razvangeorge, 2018-01-23:
Please help me to configure the camera on this NAS, i tried everything i know but i still get errors.
Thank you.

[image]

jjrojoc, 2018-01-23:
Imposible for now?

razvangeorge, 2018-01-24:
i need 2 things:
Http url and port, i don't know where to find it.

jjrojoc, 2018-01-24:
Read over hack, right now it's not implemented.
not rtsp, not http video, only in app.

shadow-1, 2018-01-28:
@razvangeorge 
@jjrojoc is absolutely correct. At this time it is only possible to use these cameras with the official Xiaomi smartphone app.

DIDIERBZH78, 2024-12-06:
hello, I'm new here and Iloock  for this software to watch my YI iot camera on my QNAS; It"s possible to get this application ?