shadow-1/yi-hack-v3 #133: 1080p Home ver 48CN will not recognize the MicroSD

elaprty, 2018-01-24:
I have successfully been able to load and stream, but the app sees my SD card as broken and wants to format it.

The moment I do that and allow the app to format the sd card, the camera will stop working and the profile I used to load the camera will also stop working.

Am I missing a step?

Now the problem is that the app sees the sd card as broken, and wants to format it, if I do that, the camera stops working. If I dont format It the camera will not save any recording. What to do in this case? I am trying to connect via ftp to see if theres a step I missed in the instructions:

> Using the Camera
> SSH Server
> The SSH server is on port 22.
> 
> Default user is root. Password is blank.
> 
> Telnet Server
> The telnet server is on port 23.
> 
> Default user is root. Password is blank.
> 
> FTP Server
> The FTP server is on port 21.
> 
> Default user is root. Password is blank.
> 
> Startup Shell Script
> On the microSD card. The following shell script is executed after the camera has booted up within a folder named yi-hack-v3:
> 
> startup.sh
> External Programs and Libraries
> Xiaomi cameras have very limited free space available for additional programs or libraries. However these can be stored on the microSD card where they can be utilised from there.
> 
> Place additional programs compiled for the camera in either of these folders on the microSD card:
> 
> yi-hack-v3\bin
> yi-hack-v3\sbin
> Place additional libraries compiled for the camera in the following folder on the microSD card:
> 
> yi-hack-v3\lib
> 

But I dont see any instructions there on what to do with the sd card so that the camera will store recordings in the sd card. Im lost.

My Camera is 48CN. 1080p Ver2

shadow-1, 2018-01-28:
@elaprty 
For the camera to store recordings to the microSD card. Try formatting the card to FAT32 or exFAT manually on your PC. It is a little unusual that formatting the card in the Xiaomi smartphone app fails.

Did you try using a different microSD card?

elaprty, 2018-01-29:
I tried reformatting it in my pc using fat32 it wont accept the sd. But it does work with exfat!!! Yay! So the step that should be added to the whole hack thing is that after the camera is synced to the app. The sd card must be taken out and reformatted to exfat. Because the first formatting will not take. Fat32 when flashing, exfat when we start using the camera. Thanks!!!