shadow-1/yi-hack-v3 #134: Possible to copy an image from camera periodically?

kb5won, 2018-01-25:
I'm wondering if it is possible to copy an image a few times per second?   I have read this thread https://github.com/shadow-1/yi-hack-v3/issues/79#issuecomment-348161594 that talks about a new image being available when motion is detected.  This is close to what I'd like to do, except I'd like a new image every second, or maybe a new image 5 times a second.

If I can copy images into a folder where zoneminder can read it, I can have zoneminder do the motion detection and alarm.

kb5won, 2018-01-25:
I just discovered the 'closed' filter and found a similar request : https://github.com/shadow-1/yi-hack-v3/issues/14

If I understand correctly, there is no way to get an image periodically.   Right?

When the RTSP stream is implemented, all my desires will be realized!  :)

jjrojoc, 2018-01-25:
Yes, everybody are waiting for that.

thearn, 2018-01-27:
Though the Yi app is very convenient, I'd like this capability from within my Home Assistant setup as well. In the meantime, want to say I appreciate the work of the developers 👍

jjrojoc, 2018-01-27:
@thearn
Please how to configure and use yi 1080p done with home assistant?
What can make home assistant?

jjrojoc, 2018-01-27:
Can home assistant record video?
I normally use ffmpeg for record rtsp streams in hls  format from cameras and nginx with hls.js for watch videos