shadow-1/yi-hack-v3 #135: Unable to connect to freevpn.ninja

NeoDeGenero, 2018-01-25:
I've attempted to connect to the site via browsers, and proxied browers, seemed the site is down.

Anyone have any luck with this?

Cheers.

[image]

bola04053, 2018-01-25:
go http://spys.one/proxys/CN/
manual type in proxy on your front page , example like this : 

random_chain
chain_len = 1
tcp_read_time_out 15000
tcp_connect_time_out 8000
[ProxyList]
socks5 223.241.214.44 1217
socks5 211.95.255.42 1080
socks5 118.89.231.84 30687
socks5 119.29.163.54 1080
socks5 61.145.213.14 1080
socks5 218.58.210.233 1080
socks5 221.209.62.170 1080
socks5 117.135.132.107 3000

NeoDeGenero, 2018-01-26:
Awesome, works great, thanks heaps!

closed 2018-01-26 by NeoDeGenero

easycor90, 2018-03-29:
[image]
I've attempted to connect to the site via browsers, and proxied browers, seemed the site is down.
Anyone know how to fix it?

Thanks you so much

williamwl1215, 2018-04-04:
hey your manual type server has all failed. 
and i try to type other servers and all test config failed. 

is there any problem or mistake i made? like type the wrong server address formate?

adrianwy, 2018-04-10:
I have the same issue, all connections failed. Any other idea?