shadow-1/yi-hack-v3 #136: Yi Dome 1080p (yi-hack-v3)can't backup video to mi wifi

vip991988, 2018-01-26:
Hi, i used yi hack v3 for Yi Dome 1080p, everything work fine but i have issue when try to backup video to mi router 3 hard disk. It say"Your router is not on support" router list or attempt to obtain information from your router has failed".
I have ant yi 720p, it backup video to my mi router hard disk normally.

vip991988, 2018-01-27:
can someone help me?

vip991988, 2018-01-27:
Please help me @shadow-1

vip991988, 2018-02-08:
who can make it, please

jjrojoc, 2018-02-08:
Yi dome 1080p don't have support for the moment.

I'm using a script: https://gist.github.com/guerrerocarlos/00554400c50e96e6327e7c5dc53882e8

Modify --> d="`dirname $f`" by d=${file%/*}
Because dirname is not in camera.
I can't make script create folders in my raspberry, so I using a Cron job and creating the folders in raspberry

vip991988, 2018-02-08:
thank you, @jjrojoc May you teach me how to backup video from Yi Dome to Mi Wifi Harddisk.

jjrojoc, 2018-02-08:
Off course, if Mi WiFi have ftp server

jjrojoc, 2018-02-08:
The problem there is a Cron in router.
I ended using a Cron for only create folders with she date like yi dome record video because I can't get make it from camera.
Proftpd in raspberry give error 550 when yi dome try to create folder.

jjrojoc, 2018-02-08:
Ups bad type.
She should be "same"

vip991988, 2018-02-09:
Thank you for your support ! I used Mi Router Gen 3, it have external hard disk storagre , FTP.
Please help me make the script.
Anw, Ftpd give error 550.Did your camera have permission to access the Mi Router Hard Disk